terjeio/ioSender #81: csv files broken (fix proposed)

Hamoudy99, 2021-01-19:
while using the csv files that describe the `error codes` / `alarm codes` / `settings codes` / `build option codes` i found that the `settings codes` csv file was semicolon-separated while the others were comma-separated. no sure why that is, but it caused the csv to not be read properly due to a few misplaced `"` characters and `;` characters that were used as delimiters in some strings. (suggest to use `|` instead of `:` for the string delimiters?)

I went through the entire `settings codes` csv file and converted the necessary semicolons to commas. so it is now comma separated with semicolons existing in strings to separate different strings in a single column (not sure what they are used for exactly, but I'm sure @terjeio knows).

turns out there's a second file with the same name (but its a .txt file) and it has similar but slightly different columns and data.
now I'm not sure which to use and which to ignore. the txt file seems to have more information per row, but it has less rows overall. it does, however, reference axis X through C (6 axis) while the .csv only references 3 axis.

I can combine both files into one, but then **there will be some missing data** (mostly just the `Data Format` and `units` columns) and there are some variations in the `Data Type` column between both files (eg. `float` and `decimal`)

i want to use the csv for my custom g-code sender. and thought id help fix it in this repo as well.

you can find the combined csv file **[here](https://github.com/terjeio/Grbl-GCode-Sender/files/5836222/setting_codes_en_US.txt)**. just change the .txt file type to .csv

terjeio, 2021-01-19:
These files are for my own sender so that I could build a rich settings UI. They are likely to be removed, see below.

The master csv files will still be [distributed with grblHAL](https://github.com/terjeio/grblHAL/tree/master/doc/csv) for use by other senders. Any format issues should thus be raised for grblHAL. I am a bit unsure if these should include more columns than provided by legacy grbl.

Since keeping these files in sync with the controller was a pain I have now (currently in the grblHAL test branch) added this information to the controller from where it can be fetched via [$-commands](https://github.com/terjeio/grblHAL/wiki/Report-extensions#code-sets-and-enumerations). Currently these are only available formatted grbl-style. I am considering adding other formatting styles such as JSON later on.

terjeio, 2021-08-21:
The latest build of grblHAL (20210821) has both settings details _and_ descriptions available via [$-commands](https://github.com/grblHAL/core/wiki/Report-extensions#code-sets-and-enumerations) (except the STM32F103xB version - there is not enough flash for the descriptions). In addition to enumerations aimed at direct sender use two new enumerations has been added that outputs settings info in the original Grbl format or grblHAL format. 

FYI the latest ioSender release has a Windows-based [configuration app](https://github.com/terjeio/ioSender/releases/download/2.0.34/Config.App.2.0.34.zip) (Windows only) that has _File_ menu entries for downloading data for the new enumerations to a file.

Note that the settings provided are dependent on the controller configuration, e.g. plugins and drivers may add settings in addition to the ones handled by the core.

closed 2021-08-21 by terjeio