terjeio/ioSender #86: GUI don't catch M5 command

calabr, 2021-02-01:
Sender Beta-8.8

terjeio, 2021-02-01:
This is a bug in the controller, it does not send back information in the `|A` element in the real-time report when the spindle is turned off. IIRC an issue has already been raised about this but it has not yet been fixed. Note that if coolant it active then it is sent and detected.

Same goes for the G5x offsets, changes are not reported back when they happens.

Both these bugs are fixed in [grblHAL](https://github.com/terjeio/grblHAL).

calabr, 2021-02-01:
According to current situation with messages,  
I propose to revert back Betta 8 behaviour and add additional popup window,  for example on the click to status indicator or behind it,  with short event log,  for example 
Time,  event,  description (helpful for alarms and errors)  
For 2-3 latest events.  
Only events,  which changing state indication. 
Looking console can be confusing for some users,  especially traffic - saving communication shortening in GRBL

terjeio, 2021-02-04:
I have managed to create a workaround for M5 by using spindle speed = 0 from the `Fs:` element. I will be a little while before I update as I am currently working on Trinamic driver interface and tuning.

terjeio, 2021-04-29:
Workaround added to latest release.

closed 2021-04-29 by terjeio