terjeio/ioSender #87: keyboard jogging isues

calabr, 2021-02-01:
Sender Beta 8.8
 GRBL 1.1h

If Jogging pannel is active - keyboard move the machine in distance mode, even if keyboard move enabled.
"Jog step" in the status bar changed with buttons on Jog window, but ignore 10mm distance.

Keyboard move try to create jog command for single move with programmed distance and speed in the "keyboard jogging" settings. Default setup use 500mm move and exeeds my machine travel. I alwais get error

terjeio, 2021-02-01:
> "Jog step" in the status bar changed with buttons on Jog window, but ignore 10mm distance.

This is per design and is for short step jogging when using the `<Ctrl>`-key modifier when jogging.
If keyboard jogging is not enabled in settings and the jog flyout is visible, and no modifiers beeing pressed, then the distance selection is respected.

> Keyboard move try to create jog command for single move with programmed distance and speed in the "keyboard jogging" settings. Default setup use 500mm move and exeeds my machine travel. I always getting error
error:15 - Jog target exceeds machine travel. Jog command has been ignored.

There is a warning about enabling full keyboard jogging in the settings tab, it is advisable to _not enable_ it due to a bug in Grbl. Grbl does not have an option of limiting jog moves either (grblHAL has). If nevertheless enabling keyboard jogging there distances shold be set short to avoid problems. Or take a chance and disable Grbl soft limits.

> What I want to get from this function - it's continous moution with defined speed and stop when I release the button.
Is it possible to implement keyboard jogging like GRBL Joystic move recomendation?

Perhaps, but I am not going to implement it due to the outsanding bug in Grbl. Switch to grblHAL for better functionality - it supports keyboard jogging limited to the machine work area (as an option) with `<Ctrl>` and `<Shift>` modifiers for selecting mode and speed.

calabr, 2021-02-01:
> Grbl does not have an option of limiting jog moves either (grblHAL has). If nevertheless enabling keyboard jogging there distances shold be set short to avoid problems. Or take a chance and disable Grbl soft limits.

GRBL soft limits restricts jogging to the machine working area.  
Switching it off - isn't good idea,  except you want to set homing position in the center of the machine and rely on hard limits. 
  My machine is very comfortable with soft limits.
  Which bug you are mention?

terjeio, 2021-02-01:
> GRBL soft limits restricts jogging to the machine working area.

... by raising an alarm if outside. This is a IMO primitive way of handling it.

grblHAL has a an option where jog moves outside the working area are cut short to be within it without raising an alarm.

> Which bug you are mention?

[This](https://github.com/gnea/grbl/issues/95). There are [other reports](https://github.com/gnea/grbl/search?q=jog+cancel&type=issues) too.

closed 2021-04-30 by terjeio