terjeio/ioSender #90: bdring port parsing error

jschoch, 2021-03-15:
the parser seems to puke on the msg "[HLP:$$ $+ $# $S $L $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H $F $E=err ~ ! ? ctrl-x]'

jschoch, 2021-03-15:
quick hacks

Grbl.cs

1756
```c#
            int i = 0;
            double stepsmm;
            do
            {
                stepsmm = GetDouble(GrblSetting.AxisSetting_XTravelResolution + i);
                
                GrblInfo.TravelResolution.Values[i++] = 1d / stepsmm;
            } while (!double.IsNaN(stepsmm) && i < GrblInfo.TravelResolution.Values.Length);

```

Process(string data)...
```c#
                string[] valuepair = data.Split('=');
                if (valuepair.Length == 2 && valuepair[1] != "")
```

jschoch, 2021-03-15:
i can do a pull request but I fear the github codebase isn't in sync with your private repo.

terjeio, 2021-03-15:
Yep, the github repo is out of sync. I'll check tomorrow for the issues you have found - IIRC the out of bounds issue is fixed already.

BTW the first production release will be out soon, got confirmation today that a probing option seems to be ok now - I have been holding back for that.

jschoch, 2021-03-15:
there is some debate on if $Nx should be reported with $$, maybe something you can coordinate with bart?

Let me know when the repo is in sync and I can see what is fixed.  Looking forward to the new release.  I've been on beta 6 for a while since 7..8 never launched without crashing.

terjeio, 2021-03-16:
> there is some debate on if $Nx should be reported with $$, maybe something you can coordinate with bart?

They should not be reported as $Nx - is this the reason for _settings_ parser failing? If so I have changed it to ignore non-numeric setting ids. IMO `$Nx` is handled outside settings for a good reason - they contain gcode (when not an empty string) that is executed for validation when set.

Out of bounds check was already added. What is the reasoning behind the empty string check? grblHAL has settings with string values and an empty string is a valid value. The added check will cause them to be ignored.

jschoch, 2021-03-16:
the empty string check was the quick hack to get it to start without crashing, i'll submit an issue on the $Nx in $$

jschoch, 2021-03-24:
this PR was accepted so this should be fixed.

closed 2021-03-24 by jschoch