terjeio/ioSender #91: probing 8.10 bdring port error 2

jschoch, 2021-03-17:
i'm testing out a laser TOF module for doing tool offsets.  I'm trying to run the probe with different work offsets so I can look at the machine coordinates and see how repeatable the laser module is.  When probing from the offset tab I get this

```
[PRB:87.504,13.055,-41.232,0.000,0.000,0.000:0]
[G54:87.504,13.055,-41.107,0.000,0.000,0.000]
[G55:0.000,0.000,-41.232,0.000,0.000,0.000]
[G56:0.000,0.000,-42.109,0.000,0.000,0.000]
[G57:0.000,0.000,0.000,0.000,0.000,0.000]
[G58:0.000,0.000,0.000,0.000,0.000,0.000]
[G59:0.000,0.000,0.000,0.000,0.000,0.000]
[G28:0.000,0.000,0.000,0.000,0.000,0.000]
[G30:0.000,0.000,0.000,0.000,0.000,0.000]
[G92:0.000,0.000,0.000,0.000,0.000,0.000]
[TLO:0.000]
[PRB:87.504,13.055,-41.232,0.000,0.000,0.000:0]
[MSG:Found]
[PRB:87.504,13.055,-42.449,0.000,0.000,0.000:1]
[MSG:Found]
[PRB:87.504,13.055,-42.378,0.000,0.000,0.000:1]
error:2 - Missing the expected G-code word value or numeric value format is not valid.
[GC:G38.2 G55 G17 G21 G90 G94 M5 M9 T0 F25 S0]
```
I get other issues when probing from the other tab which i'll post another issue for.

Regarding the module, it is connected to a pro micro which simply is setting a pin LOW when the TOF measurement is < 50mm.  That pin is connected to bart's opto isolated input board mapped to the probe pin in grbl.

terjeio, 2021-03-17:
>  When probing from the offset tab I get this

Are you probing with your own macro then?

You may also want to try the latest edge version (8.10), available for download from [here](http://www.io-engineering.com/downloads). I have made some workarounds for legacy grbl thay may help, but I could well be that there are still something I have missed related to that.

Perhaps grbl_esp32 could (or should?) address some of the shortcomings of legacy grbl such as adding some gcode states that are missing from the GC report? IIRC at least G49 and G43.x are missing. The sender also makes use of the G59.3 offset for fixture probing (as does LinuxCNC) which is not present in legacy grbl and neither in grbl_esp32 as evident from the output listed above.

terjeio, 2021-03-18:
> error:2 - Missing the expected G-code word value or numeric value format is not valid.

I wonder if this is related to an issue you had earlier (IIRC) where the Z position was null? Since you are able to compile and debug will it be possible for you to do that to find out? If so I can commit the latest version without creating a binary for it. 

I am not sure why you linked to the grbl_esp32gcode parser source. What is missing (IMO) is the `G49` and `G43.1` states from the `$G` response. For legacy grbl I have to check the `$#` response to find out if any tool offset is active.

jschoch, 2021-03-18:
yes, please upload the latest source.

I linked the esp32 grbl source references for g49 and g43.x

terjeio, 2021-03-18:
uploaded now

jschoch, 2021-03-19:
it seems to be sticking in the "controller is not responding" loop. It threw this exception

```
System.NullReferenceException: 'Object reference not set to an instance of an object.'
```

line 175 of serialstream.cs

terjeio, 2021-03-22:
Could the reason be issue #92? Tests with grbl_esp32 works as it should after the fix I made.

jschoch, 2021-03-22:
i don't think the issues are connected but i can remote debug it now.  Sorry, I haven't tested the probing again.  many yaks were shaved since then.  I think i can get to it today.

jschoch, 2021-03-24:
this is the latest build from the github repo

[image]

```
[PRB:0.000,0.000,0.000:0]
ok
Probing...
G91F100
ok
PM:ok
G38.3F100Z-5
[PRB:162.448,147.437,-109.198:1]
ok
PM:ok
G1F200Z1
ok
PM:ok
G38.3F100Z-5
[PRB:162.448,147.437,-109.196:1]
ok
PM:ok
G53G1F200Z-106.5
ok
PM:ok
G53G1F200Y147.437
ok
PM:ok
G53G1F200X162.448
ok
PM:ok
G10L2P1ZNaN
G90
Probing completed
error:2
ok
```

also the console vanishes and you have to open it twice for it to reappear.

terjeio, 2021-03-24:
This:

> G10L2P1ZNaN

Z is `NaN` as you reported earlier, I'll see if I can replicate this - likely related to incorrect handling of the legacy protocol.
Luckily I now have a CNC machine simulator for testing...

terjeio, 2021-03-25:
New version with fixes uploaded to github (source only), hopefully ok now.

grbl_esp32 sends a "Found" message at the _start_ of each probing sequence, do you know what that is good for? A leftover from debugging?

I had a lot of trouble with the esp32 board when connected via a USB hub, very erratic behaviour. When bypassed it improved a lot, some random corruption of data remains but it does not go completely haywire anymore. It even managed to crash the hub... A counterfeit USB <> UART chip to blame?

jschoch, 2021-03-25:
I have also seen issues with USB with random errors running programs, i try to use Bluetooth which seems to not have any corruption.  I'll give the update a test today.

jschoch, 2021-04-02:
it is working but some oddness.

1. it reported a number that was really way way off, 120.xx vs 48.xx, this was into G58
2. the rest of the probes worked ok.
3. the app plain crashed when clicking the settings tab for the first time.
4. the app crashed another time when clicking from the offsets tab to the main menu or the probe tab.
5. there is a very odd lag happening after probing.  It seems to be comming from bart's port.  You probe and then when it is retracting after complete it takes quite a while for the position to catch up, meaning the machine moves and is done moving but the status updates show it still moving.  This doesn't seem to happen with regular movements.
6. The default XY clearance is wrong, not sure if this is intensional but since I was simply probing Z it shouldn't use those in any case.  The error also is a bit confusing, the defaults are both 5, if i set the XY clearance to 3 or 4 it still errors because a tool diameter is being used even though i'm only probing Z and I haven't thought about tool diameter at all.

terjeio, 2021-04-05:
> it reported a number that was really way way off, 120.xx vs 48.xx, this was into G58

If selecting the Offsets tab after Probing then the offsets are not always updated as they should. Is this what you experienced?
Anyway I'll commit a fix for this bug.

> the app plain crashed when clicking the settings tab for the first time.
the app crashed another time when clicking from the offsets tab to the main menu or the probe tab.

Crashed with a message or silently exited? I have seen some silent exits that I am looking into. Possibly due to too many tab change events beeing generated.

> re is a very odd lag happening after probing. It seems to be comming from bart's port. You probe and then when it is retracting after complete it takes quite a while for the position to catch up, [...]

I'll check to see if this happens with my MSP432 test machine running grblHAL.

> The default XY clearance is wrong, not sure if this is intensional but since I was simply probing Z it shouldn't use those in any case. The error also is a bit confusing, the defaults are both 5, if [...]

I'll disable the clearance fields and validation check when only probing in the Z direction in the next commit.

jschoch, 2021-04-06:
it was a silent exit, hasn't happened since.

terjeio, 2021-04-29:
Reopen if you still have problems with the latest release.

closed 2021-04-29 by terjeio