terjeio/ioSender #92: hard coded log path

jschoch, 2021-03-21:
i setup up remote debugging after some struggles and noticed this:

Seems it is hard coded and my laptop does not have a d drive so it just crashes and dies silently.
log = new StreamWriter(@"D:\grbl.txt");

terjeio, 2021-03-21:
You can disable this by removing the RESPONSELOG  symbol from CNC Core properties, Conditional Compilation Symbols in the Build screen. I guess I should move the filename to a command line argument and add a try/catch block...

I have been busy the last few days so I have not had the time to look into the crash yet - for that I have to flash grbl_esp32. And for that I need to install the ESP32 Arduino or PlatformIO framework? AFAIKT the ESP-IDF is not usable?

jschoch, 2021-03-21:
they've moved to platformio, esp-idf no go.

terjeio, 2021-03-22:
> they've moved to platformio, esp-idf no go

What a mess, 2GB of files in my user folder - seems Espressif do not know that there is a Program folder on C:

I have moved the debug path/filename to a command line parameter, `-debugfile`, and removed the _GCode Sender.csproj.user_ file. Just committed to github.

jschoch, 2021-04-02:
seems fixed, tested fixed

closed 2021-04-02 by jschoch