terjeio/ioSender #101: Settings:Grbl Tab not showing all parameters

glct12fsl, 2021-05-16:
Hello,
i am using a grblHAL Teensy controller on a laser engraver and suddenly 
the Setting:Grbl Tab is not showing as grouped settings, but in a raw form and
also not showing some settings names.
I downloaded again ioesender and launched it, but no cure.
Also i flashed a prebuid hex from Phil Barrett configurations, but no cure.

terjeio, 2021-05-16:
Which controller build and is ENUMS available? See _Help > About_:

[image]

glct12fsl, 2021-05-16:
Hello,
this is a snapshot of iosender info.
[image]

raenji-sk, 2021-05-16:
Don't know if something changed since I flashed the Teensy about a month or two ago. I did flash the pre built hex first and did not see all options in the Iosender config. Apparently the prebuilts are a bit old.

After compiling and flashing a hex from the latest source (the new repository, with separate core and drivers) files it all started to work perfectly.

glct12fsl, 2021-05-16:
Also snapshots of the tab.
[image]

[image]
[image]
[image]

terjeio, 2021-05-16:
That is an old version from last year. Grouped (ENUMS) settings info was not available from the controller then, if not available the old style settings view is used and some may be missing. Recompile with the latest version, preferably from the [new repo](https://github.com/grblHAL/iMXRT1062).

glct12fsl, 2021-05-16:
@terjeio and @raenji-sk  i followed your suggestion and the problem solved.
Thank you.

closed 2021-05-16 by glct12fsl