terjeio/ioSender #102: M7 - mist and Arduino

raenji-sk, 2021-05-17:
Looking at the latest Edge release, it gives an error on M7 gcode - unsupported or invalid g-code.
M8 - flood works OK
Testing with Arduino UNO so don't know if it's a general error or only Arduino (8bit grbl) related. Will try with Teensy later.

[image]

terjeio, 2021-05-17:
M7 is not universally supported by controller firmware, this is why M7 (and some other commands) can be removed on load. See the _Settings: App tab_ - _GCode command stripping_.

raenji-sk, 2021-05-17:
Oh, right. I thought that the grbl shield had more coolant options. Looking at the shields I have I see only one coolant output and it is triggered by the M8 command.
Got a bit confused now that I switched to grblHAL and teensy which have many more options.

closed 2021-05-17 by raenji-sk