terjeio/ioSender #103: Running IOSender without a controller

raenji-sk, 2021-05-18:
IOSender (latest edge release) crashes when opening any gcode and no controller is attached.
Is that supposed to work like that? I think most senders work without a controller....

terjeio, 2021-05-18:
ioSender does not work without a controller - it reads settings etc. from the controller on connect so it can configure itself accordingly. I _might_ change this later but I do not regard this as a high priority issue.

raenji-sk, 2021-05-18:
Ok, no problem. It just would make some testing and debugging more convenient if it would work without a controller and perhaps act as if a proper grblHAL controller was connected. If a thing like that is possible of course.

terjeio, 2021-05-18:
> ... act as if a proper grblHAL controller was connected. If a thing like that is possible of course.

The [simulator](https://github.com/grblHAL/Simulator) could be an option then? I have compiled and run it under Linux and connected to it via a network connection. Note that it is quite resource hungry...

It should be possible to compile it for Windows but I have not tried that.

raenji-sk, 2021-05-19:
That might work. Could it be an option to run it without a controller at all with all tabs active - independent from the controller?

I tried the simulator, compiles fine in wsl2, but the comcat fake tty is not working properly. Don't have any Linux installed...
When compiling for windows it asks for a bunch of windows ".h" files. Got some from the wine repo, but it still wants more.

Sorry for the bad description, I am not really good with Linux...
Can you maybe try windows compile when you have time for it?

raenji-sk, 2021-05-20:
Can't get it to run even with network. Maybe WSL does not allow it... Always says that it can not bind socket. And if I input some crazy high number like 25639 then it does not complain about the socket but the simulator does not start. 
BTW, It works OK without any connection (network or tty) and I can access the GRBL commands in terminal.

axg20202, 2023-11-09:
I managed to get this to work on a Windows machine in case anyone wants to try this. Download the grblhal sim for windows using the webbuilder. Open a terminal window in the folder containing the sim exe. Then run the command ./grblHAL_sim.exe -n -p 23  
Launch iosneder. In the Network tab, enter port 23, and then the ip address of the machine that the sim is running on. Iosender will launch properly.