terjeio/ioSender #104: When the source code of version V2.0.33 is compiled in Visual Studio, an error will be reported

lianzaozi, 2021-05-18:
The source code of version V2.0.33 is in the vs compiler, and the compilation fails, and the source code will report an error when compiling

terjeio, 2021-05-18:
It might be due to a missing reference? The last version needs the WPF Toolkit as there is no color picker in Microsofts libraries.
Easiest is to download a precompiled binary from [here](https://github.com/xceedsoftware/wpftoolkit/releases) and unzip it to the same folder as the sender projects resides in. I will update the build instructions for this - I forgot to to that earlier.

If this does not resolve your issue then you have to post some details about the errors you get. A precompiled version is also available from [Releases](https://github.com/terjeio/ioSender/releases/tag/2.0.33) if that is of any help.

lianzaozi, 2021-05-19:
VisualStudio Error message:

CS0266 Unable to implicitly convert the type "System.Collections.Generic.IList<System.Windows.Media.Media3D.Point3D>" to "System.Windows.Media.Media3D.Point3DCollection". There is an explicit conversion (is there a lack of forced conversion?) CNC Controls Probing D:\Grbl-GCode-Sender\ioSender V2.0.33\ioSender-2.0.33 source code\ioSender-2.0.33\CNC Controls Probing\CNC Controls Probing \HeightMapControl.xaml.cs

The name "EnumValueToVisibleConverter" does not exist in the XDG0008 namespace "clr-namespace:CNC.Controls.Probing". CNC Controls Probing D:\Grbl-GCode-Sender\ioSender V2.0.33\ioSender-2.0.33 source code\ioSender-2.0.33\CNC Controls Probing\CNC Controls Probing\EdgeFinderControl.xaml

terjeio, 2021-05-19:
> CS0266 Unable to implicitly convert the type "System.Collections.Generic.IList<System.Windows.Media.Media3D.Point3D>" to "System.Windows.Media.Media3D.Point3DCollection". 

Do you have a line number for this error. I have no idea where this error could be triggered.

> The name "EnumValueToVisibleConverter" does not exist in the XDG0008 namespace "clr-namespace:CNC.Controls.Probing". 

A missing or corrupt file? The method is in [this file](https://github.com/terjeio/ioSender/blob/master/CNC%20Controls%20Probing/CNC%20Controls%20Probing/Converters.cs).

FYI I am compiling with Visual Studio 2015 Pro. Which version are you using?

lianzaozi, 2021-05-19:
I use Visual Studio2019

[image]

lianzaozi, 2021-05-19:
I put the dll files referenced by the source code in a folder and re-referenced these dll files, but I still get an error.
Can all the dll files referenced by the source code be placed in a folder in future releases? If you rely on the network to download these dll files from the Internet, sometimes the download will fail if the network is not good

terjeio, 2021-05-19:
> CS0266 Unable to implicitly convert the type "System.Collections.Generic.IList<System.Windows.Media.Media3D.Point3D>" to "System.Windows.Media.Media3D.Point3DCollection". 

So `boundary.Points` and `mapPoints.Points` are no longer of type `Point3DCollection` in VS 2019? A cast can fix that? I do not have VS 2019 installed so cannot test.

> The name "EnumValueToVisibleConverter" does not exist in the XDG0008 namespace "clr-namespace:CNC.Controls.Probing". 

It exists in the namespace, is [the file](https://github.com/terjeio/ioSender/blob/master/CNC%20Controls%20Probing/CNC%20Controls%20Probing/Converters.cs) there? Or is it due to another breaking change by Microsoft?

> Can all the dll files referenced by the source code be placed in a folder in future releases?

I am not too keen to put compiled code together with the source. What I could to is to add some info in the [Building Wiki page](https://github.com/terjeio/ioSender/wiki/Building) for how to extract the files from a release and where to place them.

lianzaozi, 2021-06-27:
You can download the software source code ofV2.0.33, open it with VS on your computer, and compile it to see if there is an error.

lianzaozi, 2021-06-27:
My operating software steps are like this:

[image]
[image]

terjeio, 2021-06-29:
I'll have to install VS2019 on a computer to see if I can solve this (it compiles with VS2015 Pro that I have on my main machine). With luck I can find time tomorrow for that, if not it will have to wait until the weekend or early next week.

terjeio, 2021-06-30:
It compiles with a fresh install of VS2019, maybe your Helix toolkit version is too old? The version I use and the source code in the latest both have the same type as the target type where you get the error. Here is the Helix toolkit [source code](https://github.com/helix-toolkit/helix-toolkit/blob/5df4339643684c1fbc9e59aac40da449ec9effd7/Source/HelixToolkit.Wpf.Shared/Visual3Ds/ScreenSpaceVisuals/ScreenSpaceVisual3D.cs#L135-L146) in question. Does your local version match?

lianzaozi, 2021-07-01:
Thank you for your reply, I replaced the file "HelixToolkit.Wpf.dll" again, and now the source code can be compiled normally.

closed 2021-07-27 by terjeio