terjeio/ioSender #113: Hope the next version of the software will improve.

lianzaozi, 2021-06-25:
1. The software will freeze when switching the interface (for example, switching from the setting interface to the display interface)
2. The function of serial port information setting is placed in the menu bar, or the setting interface
3. Add the software to follow the control panel coordinate display function

terjeio, 2021-07-04:
1. Which sender version? And which controller?
2. Not easily done as the sender configures itself on startup by requesting information from the controller. I may add dynamic reconfiguration later so it will be possible to add a menu item.
3. I do not understand this, please explain in more detail.

lianzaozi, 2021-07-07:
3. Display the axis coordinate value of the control board in real time in the software coordinate system. For example, a point can be used to represent the position of the axis in the coordinate system

lianzaozi, 2021-07-07:
1. The latest version of the software is sometimes dying when the switch interface is displayed

terjeio, 2021-07-08:
I still don't get it...

> Display the axis coordinate value of the control board in real time in the software coordinate system. For example, a point can be used to represent the position of the axis in the coordinate system

What do you mean by the "software coordinate system"? The machine coordinate or the work coordinate or something else? The work coordinate is the machine coordinate plus any offsets.

> For example, a point can be used to represent the position of the axis in the coordinate system

You mean some kind of graphical representation similar to the existing _3D View_ where the tool is displayed as a cone?

> The latest version of the software is sometimes dying when the switch interface is displayed

Switch interface is the _Settings: Grbl_ tab? I have not experienced any crashes when changing to that in a long time, but then I am only working with grblHAL based controllers. Is your controller Grbl based? And which version/build date?

lianzaozi, 2021-07-08:
Real-time display of the current axis position in 3D View

As shown in the figure below, the red point in the blue box in the figure below represents the current position of the axis

[image]

lianzaozi, 2021-07-08:
The effect is as shown in the gif file below.

[image]

terjeio, 2021-07-13:
Try the _ioSender Edge.34p2_ version found [here](http://www.io-engineering.com/downloads). You have to enable the text overlay in _Settings: App_ as it duplicates information in the main window and will add enough overhead to slow down very fast laser engravings.

[image]

An option to change the current tool cone to a red dot or a blue cross will have to wait.