terjeio/ioSender #114: Crash after M0 macro

kfmut, 2021-07-04:
Hi,

I'm doing some tests of macros functionality and noticed that program pause issued with M0 m-code results in crash right after I hit `Cycle Start`. Single issued M0 m-code also results in crash after `Cycle Start`. Verbose output on console before crash doesn't show anything interesting, just `Hold` state.

ioSender version is 2.0.33

Macro I was testing:
```
G21
G94
G90
G53 G0 Z-1
G53 X-300 Y-354
M0
G53 X-150.5 Y-1
G53 G1 Z-84 F500
```

terjeio, 2021-07-04:
Yep, that is a bug. Fixed in next release - I'll see if I can publish it today.

kfmut, 2021-07-04:
@terjeio

Great!

Further testing revealed that if ioSender is loaded with a program and macro with M0 is invoked, program is automatically started right after the macro. Will it be fixed too?

terjeio, 2021-07-04:
I think so, can you test with the [34p1 edge version](http://www.io-engineering.com/downloads)?

kfmut, 2021-07-04:
@terjeio 

Yep, it's works but `ok` response from macro commands are assigned to blocks of loaded program, but definetly no crashes in this version :-)

[image]

terjeio, 2021-08-21:
> Yep, it's works but ok responses from macro commands are assigned to blocks of loaded program, but definetly no crashes in this version

This should be fixed in the lates release.

closed 2021-08-21 by terjeio