terjeio/ioSender #117: Home button is not active at startup

kfmut, 2021-07-10:
Hi,

I just want to report a small but annoying glitch with ioSender. If $22 is set with 5(Homing enabled and Homing on startup required) `Home` button at the bottom of the screen is not active, homing could be done only via issuing the $H command. I've checked it in 2.0.33 and edge versions:

[image]

If $22 is set with just 1(Homing enabled) button is active.

```
[VER:1.1f.20210608:]
[OPT:VNS2,35,1024,3,0]
[NEWOPT:ENUMS,RT+,HOME,TC]
[FIRMWARE:grblHAL]
[NVS STORAGE:*EEPROM]
[DRIVER:STM32F411]
[DRIVER VERSION:210606]
[BOARD:Custom BlackPill STMF411]
[PLUGIN:KEYPAD v1.00]
```

```
$0=10.0
$1=25
$2=0
$3=5
$4=0
$5=0
$6=0
$7=0
$10=511
$11=0.010
$12=0.002
$13=0
$14=7
$15=0
$16=0
$17=0
$18=0
$19=0
$20=1
$21=0
$22=5
$23=3
$24=30.0
$25=1500.0
$26=250
$27=2.000
$28=0.100
$29=0.0
$30=4800.000
$31=0.000
$32=0
$33=5000.0
$34=0.0
$35=0.0
$36=100.0
$37=0
$39=1
$40=1
$43=1
$44=4
$45=3
$46=0
$50=100.0
$51=600.0
$52=3000.0
$53=0.250
$54=500.0
$55=3000.0
$62=0
$63=2
$64=0
$65=0
$80=1.000
$81=0.010
$82=0.000
$84=0.000
$85=10.000
$90=0.000
$91=0.000
$92=0.000
$95=0.000
$100=33.334
$101=66.726
$102=199.077
$110=9610.000
$111=9610.000
$112=2550.000
$120=110.000
$121=110.000
$122=110.000
$130=301.000
$131=355.000
$132=85.000
$170=0.000
$171=0.000
$172=0.000
$341=3
$342=50.0
$343=30.0
$344=120.0
$345=120.0
$347=5.0
$348=2.500
$349=25.000
```

terjeio, 2021-07-11:
This works for me when connecting to a MSP432 based controller, could be a timing issue (the control is perhaps not ready to process events when the alarm state is published from the viewmodel). 

What happens when you press _Reset_, is _Home_ enabled then?

terjeio, 2021-07-11:
A workaround/fix for this is to check _Alarm substatus_ for the _Status report options_ in the _General_ settings (or set $10 to 1535). I'll look into how I can fix this later.

kfmut, 2021-07-12:
@terjeio

> What happens when you press Reset, is Home enabled then?

It's activates for a few moments then it's greyed out again.

> A workaround/fix for this is to check Alarm substatus for the Status report options in the General settings (or set $10 to 1535). I'll look into how I can fix this later.

Yep, it's working that way, thank you!

Should I close this issue or you will close it yourself when it would be fixed?

terjeio, 2021-07-12:
I will close it as I am going to be quite busy with other work for a week or two - and this could be a tricky issue to fix without the alarm substatus available.

kfmut, 2021-07-12:
@terjeio

Got it! :-)