terjeio/ioSender #118: Save causes odd behavior after changing settings?

phil-barrett, 2021-07-17:
This is a strange one.  Bring up ioSender V2.0.33. Open the jog panel and jog X to something other than 0. Go into the Settings: Grbl tab, Change $100 (X axis travel resolution) to something different and press enter.  Then click on save.  Go back to the Grbl Tab and open the jog panel.

At this point you will see several issues. The X axis DRO has changed, the Zero all button does not zero X (it will zero all other axes though) and the zero button next to X will not work either. If you jog X, it will move to something different but not by the jog distance.

The Reset button will clear the weird state.  

Now, if you repeat the entire sequence except do not click on Save in the Settings: Grbl tab, you will get expected behavior. 

I suspect some sort of wild pointer or initialized variable so the observed behavior may be different.

terjeio, 2021-07-17:
I believe I have seen this mentioned before...

It is due to the position beeing stored as integer values (stepper steps), this is transformed to/from a "real" position via the steps/mm setting. If changing steps/mm when the step count is non-zero then this results in weird "real" positions. Grbl behaves the same, however it should be fairly easy to recalculate the steps values, but there are others that needs to be recalulated as well so it might fast get a bit complicated. A cold start followed by resetting all offsets is the best way to get everything in order after a steps/mm change?

phil-barrett, 2021-07-17:
So, save really has nothing to do with it even though I don't see any weird effects if I don't hit save?

terjeio, 2021-07-18:
I was wrong about the offsets, they are stored as "real" positions - not a good idea to comment just before going to bed...
In the coming build I have added recalculation of the relevant integer step values on a steps/mm change, at least it can't get worse that it is now. However I would still recommend a full reset after changing settings.

> So, save really has nothing to do with it even though I don't see any weird effects if I don't hit save?

If you don't hit _Save_ the controller settings are not updated. Press _Reload_ to get back the current value(s) if not saving.

phil-barrett, 2021-07-18:
In ioSender, if I go into the Settings: Grbl tab and change a setting, it appears to be accepted.  Going back to the Grbl tab and then returning to Setting: Grbl, the change is still there. But it hasn't actually been made to the controller. If I exit ioSender and power cycle the grblHAL board, the setting clearly was not accepted.  So even though Grbl settings in ioSender shows it was changed, grblHAL hasn't really had the changes made. This is a change from past versions of ioSender?

Perhaps it would be good to have a warning that a save is required in the Settings: Grbl tab?

terjeio, 2021-07-18:
> So even though Grbl settings in ioSender shows it was changed, grblHAL [edit, er the controller, could be other Grbls] hasn't really had the changes made. This is a change from past versions of ioSender?

No.

> Perhaps it would be good to have a warning that a save is required in the Settings: Grbl tab?

Yes, and perhaps auto reload on opening the tab as well. Settings could have been changed from the MDI.
Or should I save changes immediately as newer software does and remove both _Save_ and _Reload_ buttons? Or auto save on leaving the tab?

terjeio, 2021-07-18:
Can you try the latest [edge version](http://www.io-engineering.com/downloads)? I have added a warning and auto reload on opening the settings tab.

phil-barrett, 2021-07-18:
> Yes, and perhaps auto reload on opening the tab as well. Settings could have been changed from the MDI.
> Or should I save changes immediately as newer software does and remove both Save and Reload buttons? Or auto save on leaving the tab?

Hmm.  You have thought about this more than I ...

A goal of any software should be to avoid surprises.  I think what the settings are showing should be the same as what is in the controller.  Or, at least that needs to be pointed out.  Right now, if I make a change to say X axis steps/mm, it shows up on the left side of the panel yet it is not actually in the controller. There is a lot of UX debate about setting an entry immediately or waiting for the user to make an explicit commit. Personally, I prefer the first as the user does not need to remember that there are things that have not actually been set.  The idea is that the left side of the panel always shows what is in the controller.  In that case, save and restore should go away.

For what it is worth, the mouse-over tool tips are pretty clear.  I just was going too fast to read them.

I agree about reload on Settings: Grbl panel opening.  Using MDI to set is not a common thing but could lead to confusion. I noticed that if you click on the entry that you changed via MDI and then hit reload, the edit box doesn't change - shows the previous value. A very minor issue, for sure.

By the way, I love backup and restore.  I find it useful.

phil-barrett, 2021-07-18:
Passed in flight.  Yes, I will try the edge version.

phil-barrett, 2021-07-18:
Tried it.  That certainly works. And MDI updates also work. No surprises then.

closed 2021-08-21 by terjeio