terjeio/ioSender #119: The "Highlight completed cuts" option cannot be set. How can this option be set?

lianzaozi, 2021-07-18:
The "Highlight completed cuts" option cannot be set. How can this option be set?

Another problem:
Do you have plans to add imported vector graphics (dxf, svg, plt, etc.) to the software to generate gcode?

[image]

lianzaozi, 2021-07-18:
The next version can add an option that allows the user to decide whether to display or not to display the empty shift path.

terjeio, 2021-07-18:
> The "Highlight completed cuts" option cannot be set. How can this option be set?

Currently only by editing the App.config file (RenderExecuted tag). However, I have delibaretely not enabled it as this is work in progress.

> Do you have plans to add imported vector graphics (dxf, svg, plt, etc.) to the software to generate gcode?

There is rudimentary support for .plt (HPGL), enough to create files for laser cutting PCB stensils or milling PCB outlines. Note that KiCad v5 generated files that have rounded corners for compononets is currently not working, I'll try to fix that.

Excellon drill files (.drl) are also supported but requires a controller, such as grblHAL, that supports canned drill cycles.

I have no plan to add .svg or .dxf support as that is better left to a proper CAM tool as drawing objects has to be selected (and possibly closed) and toolpath types assigned to them (inside, outside or on path cuts, conventional or climb milling, pockets and what not). Adding this in a proper way would be a large project that I am not prepared to do myself - I simply do not have time for it. You can do it?

> The next version can add an option that allows the user to decide whether to display or not to display the empty shift path.

I assume you mean the rapids (G0) movements. I'll consider that, for now set the rapids color equal to the background color.