terjeio/ioSender #121: 3 D View - Drag Knife

MARIOBASZ, 2021-07-28:
New using your sender: I normally use bCNC, and despite my very rudimentary knowledge, I have been able to make my little contributions. If I can contribute something, I will be pleased. And congratulations and my respects; huge jobs with Hal.
Drag Knife generates the path on the selected code? What is the way to enable it?
On the other hand, in 3D View exclusive tab, it allows to zoom and rotate; Is it possible to move the zoomed or rotated view? And that allows it to be seen while executing gcode (maybe as a floating window?)
In 3D View inside grbl tab (between program and console) I have not been able to zoom or rotate.
Hugs

terjeio, 2021-07-29:
> Drag Knife generates the path on the selected code? What is the way to enable it?

_File > Transform > Add drag knife moves_

> On the other hand, in 3D View exclusive tab, it allows to zoom and rotate; Is it possible to move the zoomed or rotated view? And that allows it to be seen while executing gcode (maybe as a floating window)?

Yes to all (except floating window), documented [here](https://github.com/terjeio/ioSender/wiki/Usage-tips#3d-viewer). A shortcut to that page is via _Help > Usage tips_ in the sender.

MARIOBASZ, 2021-07-29:
I have not found transform file
3D View as the main screen is really very powerful. However, it does not respond to Cycle Start (Alt + R) or taking the pin to Gnd, nor can it be accessed with the file running. Ok, it is not important existing 3D View inside grbl tab.

terjeio, 2021-07-29:
I have not found transform file

It is a submenu of the File menu in the top left corner. If no file is loaded it is greyed out.

> 3D View as the main screen is really very powerful. However, it does not respond to Cycle Start (Alt + R) or taking the pin to Gnd, nor can it be accessed with the file running.

This is done on purpose, that view may get axed later. Use the 3D View tab in the Grbl tab as you have already found out.

MARIOBASZ, 2021-07-29:
Thanks. It works now.

closed 2021-07-30 by MARIOBASZ