terjeio/ioSender #126: Controller on ethernet not responding after performing a Reset/Unlock

LucaDbHub, 2021-08-12:
First of all, i apologize in advance if the subject has been already discussed but i can't find it anywhere
I'm testing a Phil Barret's board + Tennsy 4.1 and i'm having this issues, only if connected through Ethernet/Websocket.

The final project will work on a Raspberry Pi4 with (former) Rasbian,, but i also want the possibility to connect to another pc too, that's why i'm testing on Win10 at the moment.

The problem comes when a Limit switch engages and, after permormig the Reset/Unlock sequence, IoSender stops working, giving me a "Controller not responding" message. Only option is to close it and restart it.

One note: I'm gonna use a crossover cable to direct connect to the controller, maybe it could solve the issue?
Thanks

terjeio, 2021-08-12:
It seems this is due to a rather obscure threading issue in the sender as it works as it should with a serial connection.
I have uploaded an updated [edge version](http://www.io-engineering.com/downloads) with what I believe is a fix for you to try.

LucaDbHub, 2021-08-12:
> It seems this is due to a rather obscure threading issue in the sender as it works as it should with a serial connection.
> I have uploaded an updated [edge version](http://www.io-engineering.com/downloads) with what I believe is a fix for you to try.

Hi Terejo
It works fine...tested all axis limits and non problems Unlocking the machine.
Thanks a lot for the fix, much appreciated

terjeio, 2021-08-21:
New release is out with the fix.

closed 2021-08-21 by terjeio