terjeio/ioSender #128: Trimatic tab not showing up

Cncninja, 2021-09-07:
I have 2209s installed and it shows the trinamic 0.05 plug-in loaded via the console but no tab is loading in the GUI. I'm sure this is an operator error but if you could point me in the right direction... thank you so much for all your hard work!!

Cncninja, 2021-09-07:
i just relized the 2209 is uart not spi... So i guess this will not work with grblhall and a skr1.4turbo ill look at buying some 2130s

terjeio, 2021-09-07:
You must enable at least one Trinamic driver with [$338](https://github.com/grblHAL/Plugins_motor#settings) before it shows up.

> i just relized the 2209 is uart not spi... So i guess this will not work with grblhall and a skr1.4turbo ill look at buying some 2130s

From the SKR schematic it seems UART mode drivers are not supported. Either you'll have to run them in non-UART mode or switch to compatible SPI mode drivers.

Cncninja, 2021-09-07:
the skr1.4 does support uart mode drivers.  but not sure grblhal with the current lpc17XX driver does

Cncninja, 2021-09-07:
[image]
[image]
[image]

[image]

terjeio, 2021-09-07:
>  but not sure grblhal with the current lpc17XX driver does

It does not - you'll have to add low-level UART driver code for that. Likely that has to be a software UART for each axis? I've got no idea of how much work that would be... Perhaps Marlin has code that can be transplanted?

Cncninja, 2021-09-07:
Thank you for the reply and your hard work.. i just got the wrong drivers. ill look at getting some 2130's or just running these in step/dir mode

Thank you again

terjeio, 2022-06-15:
FYI soft UART support was added to the LPC17xx driver in january.

closed 2022-06-15 by terjeio