terjeio/ioSender #132: resize windows

Cncninja, 2021-09-20:
Is there a way to resize the windows and tabs, I'm using a 12in 1920 touch screen, the tabs and windows are a bit small even with low screen resolutions.  I see the GUI can shift colors and a few things in the config tab. and i looked in the config file to see if there is a res pram.

terjeio, 2021-09-20:
The main window can be resized, the Help > About and Camera dialogs can not. There is a checkbox in Settings > App for restoring the last window size on startup.

The tab buttons can be made a bit larger I guess, at least in height.

Cncninja, 2021-09-20:
how would i do it?

Thank you, do you have a Patron?

terjeio, 2021-09-20:
> how would i do it?

You need to download the source and compile with Visual Studio.

I guess the tab buttons size can be set on this line (I am not sure):

https://github.com/terjeio/ioSender/blob/7f454de782bfcc65378deb2bc5e0aa4efe405740/GCode%20Sender/GCode%20Sender/MainWindow.xaml#L105

If it is a matter of increasing size a bit I can do that and upload a edge build for you to try. But hard to tell what size you want... Any hint?

> do you have a Patron?

Nope, I am quite old so I have a pension 😀

closed 2023-04-29 by terjeio