terjeio/ioSender #133: Req: Macro support and 3 point circle center finder

5ocworkshop, 2021-10-08:
Hi Terje,

I started using some great macros in CNCjs that I'm hoping you might be able to incorporate in to IOSender.

They are found here https://github.com/chuyskywalker/cncjs-macros and they find the center of the circle by probing 3 points on the circle rather than doing the four point, side to side probe.

The advantage being that you can position your bit in the upper right quadrant of the circle on the corner finder (assuming a lower left corner you are trying to find) and it will not push the corner finding block off the workpiece at all.  So you don't need to brace it during probing.  It pushes it along axes that hold it to the work.  It is also incredibly fast and can work on even large circles if you are probing a large circular work piece.

A photo of the corner finder I use:

[https://www.dropbox.com/s/ajv09mff0sdyt45/20191028_160226%20-%20Shopbot%20XYZ%20Finder%20Block%20First%20Sample%20Xometry%20%28Large%29.jpg?dl=0](url)

Also, I'm wondering about general macro support in IOSender.  I'd love to be able to write macros that have access to the key variables of the system as well as more traditional programming like the math used to do the center finding in the macro above.  Do you have any plans to add macro support to IOSender?

Thanks.

terjeio, 2021-10-10:
>  Do you have any plans to add macro support to IOSender?

There is already and if you [enable expression support](https://github.com/grblHAL/core/blob/master/changelog.md) in the controller you have access to [system parameters](http://www.linuxcnc.org/docs/html/gcode/overview.html#_parameters) and you can define your own, both numbered and named. Expression support is available from build 20210928.

Later I want to add parameter and expression support locally in ioSender as well, for that I have to figure out how to keep the controller and sender in sync...

5ocworkshop, 2021-10-10:
Great, I'll take a look, thank you.

What would you recommend implementing the math on the 3pt circle center finding?  It would be nice if this could all be done sender side.

5ocworkshop, 2021-10-10:
I think it is positive to support the structure that LinuxCNC supports, but also that should be a subset of the macro capability as it is limited and a model from the distant past.  Having a high level of programmatic and variable and message capability in the macro module of the sender will allow for quite a bit of sender side innovation and automation and is one of the big benefits of grbl over LinuxCNC.  I realize this is a big area and you are juggling a lot of priorities, just wanted to share the thought for consideration as time allows.

terjeio, 2021-10-10:
>  but also that should be a subset of the macro capability as it is limited and a model from the distant past

Well, it is part of the NIST RS247/NGC specification with some extensions (named parameters and some operators). What is missing is branching (`if ... then ... else`) and subroutines, is that supported by cncjs? If not the cncjs macros can be implemented with standard NIST syntax, at least those you linked to above can (IMO).
BTW is there a page somewhere that describes the cncjs macro syntax, predefined parameters and operators? I cannot find any after a brief search.

Having parameter and expression support in the controller makes this available to all senders, both in regular gcode programs and macros - I would say this is a plus. The downside to having it in the controller is when parameter values are used for motion that is to be rendered - their values are not known until the program is run unless precalculated. To work around this I plan implement the same syntax in the sender parser and substitute parameters and expressions with real values where possible. At least branching should be implemented too so that appropriate action can be taken, e.g. when a probe operation fails. This means that some parameter values has to be fetched from the controller at run-time.

>  Having a high level of programmatic and variable and message capability in the macro module of the sender will allow for quite a bit of sender side innovation and automation and is one of the big benefits of grbl over LinuxCNC.

What kind of macro language do you suggest will be appropriate? I know that Mach3 macros can be programmed in VB-script, is it something like that you are envisaging?