terjeio/ioSender #135: Keyboard jog commands trigger soft limits when using continuous mode but work fine in step mode

timm052, 2021-10-16:
For some reason when using the keyboard to jog the soft limits are triggered well before their actual set value is reached the position this occurs at changes each reset and home cycle. And when holding down CTRL to use step jogging the machine will continue to move and not give the error. I'm using GRBL_ESP32 so it may be related to the jogging issues with that firmware but i did not see this issue mentioned under limitations in the wiki. I've had a similar issue when homing occasionally where the y axis will not home correctly and gives a soft limits error about halfway to the homing switch. I do not think this is an issue of lost steps as the machine still knows it's correct absolute location.

terjeio, 2021-10-16:
> For some reason when using the keyboard to jog the soft limits are triggered well before their actual set value is reached the position this occurs at changes each reset and home cycle.

I guess this is a sender issue (or feature?) as keyboard jogging sends long distance jog commands that may exceed soft limits. So jogging with soft limits enabled will fail unless jog distances are significantly reduced and jogging is kept within machine limits.

> And when holding down CTRL to use step jogging the machine will continue to move and not give the error.

This is strange, a single jog short distance jog command is sent so it should not happen. If you send a short jog command from the MDI does that fail too? Also, be aware that a jog cancel command is not sent on key up in step mode - the distance traveled can not be cut short.

>  I'm using GRBL_ESP32 so it may be related to the jogging issues with that firmware but i did not see this issue mentioned under limitations in the wiki.

Are you able to try the [grblHAL driver](https://github.com/grblHAL/ESP32)?
I believe I have fixed the jog cancel issue present in legacy Grbl as I have not seen (or heard about) any jog issues in a very long time. grblHAL also has a new setting to limit jog motions to within machine limits ($40=1) that allows soft limits to be enabled.

terjeio, 2021-10-18:
> I plan on trying grblHAL but I'm going to have a do a custom board profile for it so that might take a while to get my head around

Creating a custom profile should be easy if it is a board without any fancy stuff going on such as ganged axes that are currently not supported by the driver. Just copy an existing map file, e.g. _bdring_v4_map.h_, to _my_machine_map.h_ and modify the pin numbers to match your board. Then enable it in _CMakeLists.txt_ by setting `BOARD_MY_MACHINE` to `On` and `BOARD_BDRING_V4` to `Off`. Note that compiling with the Arduino IDE is not supported - this may be the biggest obstacle?

 > ...any chance of adding an option to the keyboard to permanently toggle between step and smooth jogging when using keyboard commands...

I'll have to think about that a bit - maybe it will be better to limit jog distances to be within machine limits in the sender?

closed 2021-11-10 by timm052