terjeio/ioSender #144: Keyboard Jogging wrong distance calculation.

calabr, 2021-11-01:
Keyboard jogging send wrong command for X axis on my machine
?<Idle|MPos:-341.000,-0.600,-0.500|Bf:15,128|FS:0,0>
…**$J=G91G21X682.5F300**
error:15
   _Machine travel is 342 mm - calculated twice_
?<Idle|MPos:-341.000,-0.600,-0.500|Bf:15,128|FS:0,0|Ov:100,100,100>
…?<Idle|MPos:-341.000,-0.600,-0.500|Bf:15,128|FS:0,0>
…**$J=G91G21X--341F300**
error:2
   _this probable caused by text operation to add minus sign._

Y and Z axis calculated normally.
The difference is - X axis home switch position is -342mm, Y and Z is at 0
$23=1 - X ome search negative direction

 calculation formula for jog travel can be:
**+ move:**
 $J=G53G90G21X-0.5
**- move**
$J=G53G90G21X(0.5 - MaxTravel)

for ralative move mode
  +move= -CurMpos -Gap
  -move= -MaxTravel -CurMpos +Gap
 
IoSender 2.0.35 
machine GRBL Mega238 ; 1.1h.20190825

; [OPT:V,15,128]
;
$N0=
$N1=
$0=5
$1=100
$2=0
$3=7
$4=0
$5=0
$6=0
$10=3
$11=0.010
$12=0.002
$13=0
$20=1
$21=0
$22=1
$23=1
$24=25.000
$25=1000.000
$26=20
$27=1.000
$30=12000
$31=1
$32=0
$100=400.000
$101=400.000
$102=400.000
$110=4000.000
$111=4000.000
$112=4000.000
$120=400.000
$121=400.000
$122=400.000
$130=342.000
$131=244.000
$132=108.000

terjeio, 2021-11-03:
Please try the [latest edge version](http://www.io-engineering.com/downloads) and report back.

Bert911, 2021-11-17:
Hello guys,

Had the same problem as @calabr but the new edge version did the trick.

Thanks for the awesome CnC Controll-Software.

calabr, 2021-11-18:
I confirm the edge version 36p10b fixes this issue.
During test run I found some instability in jogging function.
After some keyboard jogging program stopped to send jogging commands from keyboard and from Jog window butons.
Jog window cannot be closed by "x" button, but hide if I click Jog tab.
 Window seems to be panted on top, but input focus is still on main window. Invisible controlls from main window cam react on mouse clicks.
[image]
after pressing Reset button brogramm sumtimes starting to work properly, but sometimes hang with exception.

terjeio, 2021-11-19:
@Bert911 Thanks for feedback!

@calabr This is odd, I've never seen this happening before. Which controller firmware are you using?

calabr, 2021-11-19:
I'm using clasic GRBL 1.1h on Arduino Uno.
Looks like this icssue ralated to bug with jog cancel in GRBL.
 Is it any binary GRBLhal firmware available for BluePill board? 64K flash - I kan try to test with GGRBLhal.
I'm trying to use game console as joystic and it's sometimes generates series of keypress events in short period. After such spike GRBL sometimes stopped to respond and show input buffer exhaused. problems with jog window correlates with it. 
Also with clasic GRBL if SHIFT key unpressed just before arrow key, machine often keep going with slow speed even after arrow key unpressed.
 After shift key unpressed ioSender didn't send additional jog stop. command.
Probable solution is - do not send next jog command untill state become idle and buffer is empty plus repeat jog-stop if we send stop, but received <Jog> state report from GRBL

?<Idle|MPos:-167.878,-243.000,-13.395|Bf:15,128|FS:0,0>
…$J=G53G21X-1F2000
ok
?<Jog|MPos:-167.745,-243.000,-13.395|Bf:14,128|FS:1680,0>
?<Jog|MPos:-154.773,-243.000,-13.395|Bf:14,128|FS:2000,0>
?<Jog|MPos:-141.458,-243.000,-13.395|Bf:14,128|FS:2000,0>
?<Jog|MPos:-127.808,-243.000,-13.395|Bf:14,128|FS:2000,0|WCO:-341.000,-243.000,-30.000>
?<Jog|MPos:-114.488,-243.000,-13.395|Bf:14,128|FS:2000,0>
…$J=G53G21X-1F200
…?<Idle|MPos:-106.880,-243.000,-13.395|Bf:15,128|FS:0,0|Ov:100,100,100>
ok
?<Jog|MPos:-105.735,-243.000,-13.395|Bf:14,128|FS:200,0>
?<Jog|MPos:-104.403,-243.000,-13.395|Bf:14,128|FS:200,0>
?<Jog|MPos:-103.035,-243.000,-13.395|Bf:14,128|FS:200,0>
?<Jog|MPos:-101.703,-243.000,-13.395|Bf:14,128|FS:200,0>

terjeio, 2021-11-19:
> Is it any binary GRBLhal firmware available for BluePill board? 64K flash - I kan try to test with GGRBLhal.

Yes, but only for 128K versions. The later Blackpills (F303, F411) are a lot more capable.

> GRBL sometimes stopped to respond and show input buffer exhaused. problems with jog window correlates with it.
Also with clasic GRBL if SHIFT key unpressed just before arrow key, machine often keep going with slow speed even after arrow key unpressed.

This is due to the Grbl jogging bug.

> Probable solution is - do not send next jog command untill state become idle and buffer is empty plus repeat jog-stop if we send stop, but received state report from GRBL

The solution is to fix Grbl... I am not going to try to add a workaround for this bug in the sender.

FYI I have [patched](https://github.com/terjeio/grbl-Mega/commit/ed8d64433ae2b027d90911e26557f9a56c07b06e) the Mega version, the added code is mainly inside `#ifdef SERIAL_CANCEL .... #endif` blocks. It is not much code and could be easily backported to the basic version, flash space permitting.
Feel free to copy this code and submit it as a PR if you backport it and it is working with the Nano - but I guess it will not be merged as the Grbl project seems to be completely dead by now.

calabr, 2021-11-21:
> FYI I have patched the Mega version, the added code is mainly inside #ifdef SERIAL_CANCEL .... #endif blocks. It is not much code and could be easily backported to the basic version, flash space permitting.

I'm followed your suggestion and implement this patch for Mega328. It's works! 
**Thanks alot for such  usful solution!!!**

I didn't test it with real CNC yet - only Arduino without motors. During my tests ioSender sometimes hang, but seems it's isn't related with jogging.
 After stress-test with jog, start and stop programm, hold, resers etc. ioSender trying to execute "$I+" command, receiving errors and stopped to communicate with controller. 
`?<Idle|MPos:-232.383,-85.498,-67.135|Bf:15,128|FS:0,0>
?<Idle|MPos:-232.383,-85.498,-67.135|Bf:15,128|FS:0,0>
!?<Hold:0|MPos:-232.383,-85.498,-67.135|Bf:15,128|FS:0,0>
$I+
$I+
$I+
$I+
$I+
.
Grbl 1.1h ['$' for help]
‡‡?<Idle|MPos:-232.383,-85.498,-67.135|Bf:15,128|FS:0,0|WCO:-292.558,-197.550,-66.000>
?<Idle|MPos:-232.383,-85.498,-67.135|Bf:15,128|FS:0,0|Ov:100,100,100>
?<Idle|MPos:-232.383,-85.498,-67.135|Bf:15,128|FS:0,0>
$I+
error:3
$I+
error:3
$I+
error:3
$I+
error:3
$I+
error:3
$X
ok
.
`
I didnt find exact recipt to reproduce it yet.

terjeio, 2021-11-21:
`$I+` should not be sent when the the controller is not grblHAL. I'll see if I can find where in the code - perhaps tomorrow.

terjeio, 2021-11-22:
What is the [build date](https://github.com/gnea/grbl/blob/bfb67f0c7963fe3ce4aaf8a97f9009ea5a8db36e/grbl/grbl.h#L26) of your controller? There is a check for that to determine whether to use `$I+` instead of `$I` when querying for info.

`bool getExtended = ExtendedProtocol && Build >= 20201109;`

If newer can you set it back and try again? FYI none of the official builds has a build date later or equal to 20201109.

calabr, 2021-11-22:
Yes.  That is the issue. 
I'm rebuilt project with your patch for jog cancel and set the  current  build date and version 1.1ha
  Looks like date isn't very reliable way. 
  What is the criteria to detect grblHAL board? 
  I want to change board reply to emulate HAL for translation checking

terjeio, 2021-11-22:
> What is the criteria to detect grblHAL board?
I want to change board reply to emulate HAL for translation checking

Then you have quite a task ahead of you - when the controller is grblHAL based many of the new features are taken advantage of, more info in the wiki:

https://github.com/grblHAL/core/wiki/For-sender-developers
https://github.com/grblHAL/core/wiki/Report-extensions#controller-information-extensions

You may try the [simulator](https://github.com/grblHAL/Simulator) instead. Note that I have only compiled it for Linux, Windows is possible but I do not have my machine set up for that. The Linux version can be made accessible for ioSender via a Telnet connection.

calabr, 2021-11-28:
additional test with sender 2.0.36.p11
 All works like it should.
 Additionally Numpad6 and Numpad4 changing step for keyboard jogging, like for Jog window - it's very useful.

closed 2021-11-28 by calabr