terjeio/ioSender #145: Localization/translation

calabr, 2021-11-02:
Russian translation proposal for ioSender 2.0.35
[translation_ru-RU.txt](https://github.com/terjeio/ioSender/files/7456984/translation_ru-RU.txt)

terjeio, 2021-11-02:
Thanks.

See [this comment](https://github.com/terjeio/ioSender/discussions/140#discussioncomment-1574559) for news.
Updated files for translation can be found in the _Locale_ subfolder in the download.

calabr, 2021-12-01:
ru_RU updated for 2.0.36p13
[translation_ru-RU.txt](https://github.com/terjeio/ioSender/files/7637627/translation_ru-RU.txt)

terjeio, 2021-12-02:
Thanks, a new [edge version](http://www.io-engineering.com/downloads) up (15) with this added including an attempt at improving handling of the job buttons for legacy Grbl.

calabr, 2022-04-10:
ru-RU translation updated to ioSender Edge 2.0.38p1
[translation.txt](https://github.com/terjeio/ioSender/files/8460049/translation.txt)

terjeio, 2022-04-20:
Thanks, now added to [latest edge version](http://www.io-engineering.com/downloads).

calabr, 2022-05-17:
Updated to 2.0.38p
[translation.txt](https://github.com/terjeio/ioSender/files/8711921/translation.txt)
3