terjeio/ioSender #149: Crashing on Launch

timm052, 2021-11-10:
I'm having issues with this on an old laptop running windows 10 IOsender will start and ask if it can create a new config file but then it silently crashes without opening the main window the laptop is a fairly old AMD machine from the Windows 7 era but is fully up to date so I'm not sure what's going on. it's a fresh instal so maybe I'm missing a dependency is there any easy way of getting a crash log so I can work out whats going on?

terjeio, 2021-11-10:
Can you try the [edge version](http://www.io-engineering.com/downloads)? Note that there is a new algorithm for checking soft limits in this that needs verification.

You may find the reason for the crash in the Windows event log.

terjeio, 2021-11-10:
IIRC you are using a Grbl_ESP32 controller, ESP32 often outputs a lot of "noise" on startup and takes a fairly long time to boot. You may try to increase the sender reset delay in its _App.config_, default is 2 seconds:

`  <ResetDelay>2000</ResetDelay> `

timm052, 2021-11-10:
Same issue on the edge version, based on event viewer it looks like it's some occurring when creating the port selection window.

`Application: ioSender.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.ArgumentOutOfRangeException
   at System.ThrowHelper.ThrowArgumentOutOfRangeException(System.ExceptionArgument, System.ExceptionResource)
   at System.Collections.Generic.List`1[[System.__Canon, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]].get_Item(Int32)
   at CNC.Core.SerialPorts.Refresh()
   at CNC.Core.SerialPorts..ctor()
   at CNC.Controls.PortProperties..ctor()
   at CNC.Controls.PortDialog..ctor()
   at CNC.Controls.AppConfig.SetupAndOpen(System.String, CNC.Core.GrblViewModel, System.Windows.Threading.Dispatcher)
   at GCode_Sender.MainWindow..ctor()

Exception Info: System.Windows.Markup.XamlParseException
   at System.Windows.Markup.WpfXamlLoader.Load(System.Xaml.XamlReader, System.Xaml.IXamlObjectWriterFactory, Boolean, System.Object, System.Xaml.XamlObjectWriterSettings, System.Uri)
   at System.Windows.Markup.WpfXamlLoader.LoadBaml(System.Xaml.XamlReader, Boolean, System.Object, System.Xaml.Permissions.XamlAccessLevel, System.Uri)
   at System.Windows.Markup.XamlReader.LoadBaml(System.IO.Stream, System.Windows.Markup.ParserContext, System.Object, Boolean)
   at System.Windows.Application.LoadBamlStreamWithSyncInfo(System.IO.Stream, System.Windows.Markup.ParserContext)
   at System.Windows.Application.LoadComponent(System.Uri, Boolean)
   at System.Windows.Application.DoStartup()
   at System.Windows.Application.<.ctor>b__1_0(System.Object)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(System.Delegate, System.Object, Int32)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(System.Object, System.Delegate, System.Object, Int32, System.Delegate)
   at System.Windows.Threading.DispatcherOperation.InvokeImpl()
   at System.Windows.Threading.DispatcherOperation.InvokeInSecurityContext(System.Object)
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at MS.Internal.CulturePreservingExecutionContext.Run(MS.Internal.CulturePreservingExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Windows.Threading.DispatcherOperation.Invoke()
   at System.Windows.Threading.Dispatcher.ProcessQueue()
   at System.Windows.Threading.Dispatcher.WndProcHook(IntPtr, Int32, IntPtr, IntPtr, Boolean ByRef)
   at MS.Win32.HwndWrapper.WndProc(IntPtr, Int32, IntPtr, IntPtr, Boolean ByRef)
   at MS.Win32.HwndSubclass.DispatcherCallbackOperation(System.Object)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(System.Delegate, System.Object, Int32)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(System.Object, System.Delegate, System.Object, Int32, System.Delegate)
   at System.Windows.Threading.Dispatcher.LegacyInvokeImpl(System.Windows.Threading.DispatcherPriority, System.TimeSpan, System.Delegate, System.Object, Int32)
   at MS.Win32.HwndSubclass.SubclassWndProc(IntPtr, Int32, IntPtr, IntPtr)
   at MS.Win32.UnsafeNativeMethods.DispatchMessage(System.Windows.Interop.MSG ByRef)
   at System.Windows.Threading.Dispatcher.PushFrameImpl(System.Windows.Threading.DispatcherFrame)
   at System.Windows.Threading.Dispatcher.PushFrame(System.Windows.Threading.DispatcherFrame)
   at System.Windows.Application.RunDispatcher(System.Object)
   at System.Windows.Application.RunInternal(System.Windows.Window)
   at System.Windows.Application.Run(System.Windows.Window)
   at GCode_Sender.App.Main()
`

timm052, 2021-11-10:
hmm yeah i was setting it up upstairs not connected to the controller weird given my desktop pc can run it fine with no controller attached. i'll try using it connected in the morning.

terjeio, 2021-11-10:
Desktop PCs always has COM1 available even if there is no hardware available for it?

I'll add a check for number of ports available in the next build. The Port drop-down will show up empty and clicking OK will just close the sender down. You will still be able to connect via a network connection if the controller supports it.

infamous-panda, 2021-11-13:
Having a similar issue but the controller is connected. However I do not get far enough that I get to select ports. Event logs below

// Information

Fault bucket 1184787194649956434, type 5
Event Name: CLR20r3
Response: Not available
Cab Id: 0

Problem signature:
P1: ioSender.exe
P2: 2.0.35.0
P3: 618a0b30
P4: CNC.Core
P5: 2.0.36.0
P6: 618a0b2a
P7: 48c
P8: 94
P9: N3CTRYE2KN3C34SGL4ZQYRBFTE4M13NB
P10: 

Attached files:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER783.tmp.dmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER949.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER96A.tmp.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER968.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER9A7.tmp.txt

These files may be available here:
\\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_ioSender.exe_b23ff906614b18c31583a5df9acc0d072ffc43f_b061ab13_e5f69778-5807-4f28-9fa7-1bb8d08504d4

Analysis symbol: 
Rechecking for solution: 0
Report Id: 7db3e963-e58f-406b-91dd-e41ff188e2d5
Report Status: 268435456
Hashed bucket: d995224f46d3d921e07135ad1f8ad052
Cab Guid: 0

// Error

Faulting application name: ioSender.exe, version: 2.0.35.0, time stamp: 0x618a0b30
Faulting module name: KERNELBASE.dll, version: 10.0.19041.1288, time stamp: 0x3e55bd0b
Exception code: 0xe0434352
Fault offset: 0x0012b5b2
Faulting process id: 0x3898
Faulting application start time: 0x01d7d8c2ffcc298b
Faulting application path: C:\Users\jonas\Desktop\ioSender Edge.36p9\ioSender Edge\ioSender.exe
Faulting module path: C:\WINDOWS\System32\KERNELBASE.dll
Report Id: 7db3e963-e58f-406b-91dd-e41ff188e2d5
Faulting package full name: 
Faulting package-relative application ID: 

// Error

Application: ioSender.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.NullReferenceException
   at CNC.Core.SerialPorts.Refresh()
   at CNC.Core.SerialPorts..ctor()
   at CNC.Controls.PortProperties..ctor()
   at CNC.Controls.PortDialog..ctor()
   at CNC.Controls.AppConfig.SetupAndOpen(System.String, CNC.Core.GrblViewModel, System.Windows.Threading.Dispatcher)
   at GCode_Sender.MainWindow..ctor()

Exception Info: System.Reflection.TargetInvocationException
   at System.RuntimeTypeHandle.CreateInstance(System.RuntimeType, Boolean, Boolean, Boolean ByRef, System.RuntimeMethodHandleInternal ByRef, Boolean ByRef)
   at System.RuntimeType.CreateInstanceSlow(Boolean, Boolean, Boolean, System.Threading.StackCrawlMark ByRef)
   at System.RuntimeType.CreateInstanceDefaultCtor(Boolean, Boolean, Boolean, System.Threading.StackCrawlMark ByRef)
   at System.Activator.CreateInstance(System.Type, Boolean)
[... 19 lines cut ...]
   at System.Windows.Application.<.ctor>b__1_0(System.Object)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(System.Delegate, System.Object, Int32)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(System.Object, System.Delegate, System.Object, Int32, System.Delegate)
   at System.Windows.Threading.DispatcherOperation.InvokeImpl()
   at System.Windows.Threading.DispatcherOperation.InvokeInSecurityContext(System.Object)
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at MS.Internal.CulturePreservingExecutionContext.Run(MS.Internal.CulturePreservingExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Windows.Threading.DispatcherOperation.Invoke()
   at System.Windows.Threading.Dispatcher.ProcessQueue()
   at System.Windows.Threading.Dispatcher.WndProcHook(IntPtr, Int32, IntPtr, IntPtr, Boolean ByRef)
   at MS.Win32.HwndWrapper.WndProc(IntPtr, Int32, IntPtr, IntPtr, Boolean ByRef)
   at MS.Win32.HwndSubclass.DispatcherCallbackOperation(System.Object)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(System.Delegate, System.Object, Int32)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(System.Object, System.Delegate, System.Object, Int32, System.Delegate)
   at System.Windows.Threading.Dispatcher.LegacyInvokeImpl(System.Windows.Threading.DispatcherPriority, System.TimeSpan, System.Delegate, System.Object, Int32)
   at MS.Win32.HwndSubclass.SubclassWndProc(IntPtr, Int32, IntPtr, IntPtr)
   at MS.Win32.UnsafeNativeMethods.DispatchMessage(System.Windows.Interop.MSG ByRef)
   at System.Windows.Threading.Dispatcher.PushFrameImpl(System.Windows.Threading.DispatcherFrame)
   at System.Windows.Threading.Dispatcher.PushFrame(System.Windows.Threading.DispatcherFrame)
   at System.Windows.Application.RunDispatcher(System.Object)
   at System.Windows.Application.RunInternal(System.Windows.Window)
   at System.Windows.Application.Run(System.Windows.Window)
   at GCode_Sender.App.Main()

terjeio, 2021-11-13:
Thanks, looks like it is related to #149.
I have uploaded a new [edge version](http://www.io-engineering.com/downloads) I hope works - if not report back.

This version uses FTP to upload to from  the _SD Card_ tab when available.

infamous-panda, 2021-11-13:
> Thanks, looks like it is related to #149. I have uploaded a new [edge version](http://www.io-engineering.com/downloads) I hope works - if not report back.
> 
> This version uses FTP to upload to the card from the _SD Card_ tab when available.

This one launches and FTP upload works!

I can only connect through ethernet, though. At startup it only lets me select COM1 which is not where my controller is.

terjeio, 2021-11-13:
Hmm, this means that the loop enumerating the pins crashes after the first iteration.
I added port names recently, somehow that is causing the crash for you. Can you post the list of available ports from the Device Manager so I can check if there is something I do not expect in that?
Here is mine:
[image]

terjeio, 2021-11-13:
I have uploaded a new [edge version](http://www.io-engineering.com/downloads), hopefully this adds ports that has no extended  name or a name that triggers the exception that stops enumeration.

infamous-panda, 2021-11-13:
> Hmm, this means that the loop enumerating the pins crashes after the first iteration. I added port names recently, somehow that is causing the crash for you. Can you post the list of available ports [...]

[image]

COM6 Right now. Sometimes its COM3 if I connect it via a HUB

infamous-panda, 2021-11-13:
> I have uploaded a new [edge version](http://www.io-engineering.com/downloads), hopefully this adds ports that has no extended name or a name that triggers the exception that stops enumeration.

That one seems to work for me!

closed 2023-04-29 by terjeio