terjeio/ioSender #156: Trouble running new version

jappiemike, 2021-11-14:
Hi, every time I install a new version of iosender I have to go through the folder and unblock the files. I'n not a regular windows user so I'm not sure if this is normal or I'm doing something stupid. Any help would be welcome.
Cheers
Mike

terjeio, 2021-11-14:
This could be due to the code beeing unsigned? I could get a certificate and sign it but last time I checked it was not cheap.

phil-barrett, 2021-11-14:
Yes, it's about the code not being signed.  It's a one time thing.  For most users, it isn't much of a burden though if you are tracking lots of changes, it can be annoying.  Think of it as the price of free. I see no reason for Terje to incur a cost here.

jappiemike, 2021-11-15:
Thanks for the replies, as I said as a inexperienced windows user I wasn't sure if this was normal. I'm not fussed about doing a small bit of work compared to the massive efforts you have put in.
Cheers
Mike

closed 2021-12-03 by jappiemike