terjeio/ioSender #161: Feed hold handling issue

calabr, 2021-11-22:
sender 2.0.36p10
machine GRBL 1.1h

after next sequeence programm execution ctopped:
 1) Cycle Start
2) Feed Hold
3) stop
4) Cycle Start again
  Sender started to send program, but request initial information in the same time
like $I $G $+
which return error and programm execution freeze.
  errormessage description cometimes appear on status bar, but cannot be recognized as show-stopper - cosmetic issue

During programm execution Stop button acts first time as "Pause" button, than second press stopped execution. 
Burron caption should e changed to reflect actual action.
After program stopped Spindle still run

; 
; 1.1ha.20201101
; [OPT:V,15,128]
;
$N0=
$N1=
; 0 - Step pulse time
$0=5
; 1 - Step idle delay
$1=100
; 2 - Step pulse invert
$2=0
; 3 - Step direction invert
$3=7
; 4 - Invert step enable pin
$4=0
; 5 - Invert limit pins
$5=0
; 6 - Invert probe pin
$6=0
; 10 - Status report options
$10=3
; 11 - Junction deviation
$11=0.010
; 12 - Arc tolerance
$12=0.002
; 13 - Report in inches
$13=0
; 20 - Soft limits enable
$20=1
; 21 - Hard limits enable
$21=0
; 22 - Homing enable
$22=1
; 23 - Homing direction invert
$23=1
; 24 - Homing locate feed rate
$24=25.000
; 25 - Homing search seek rate
$25=1000.000
; 26 - Homing switch debounce delay
$26=20
; 27 - Homing switch pull-off distance
$27=1.000
; 30 - Maximum spindle speed
$30=12000
; 31 - Minimum spindle speed
$31=1
; 32 - Laser mode
$32=0
; 100 - X-axis travel resolution
$100=400.000
; 101 - Y-axis travel resolution
$101=400.000
; 102 - Z-axis travel resolution
$102=400.000
; 110 - X-axis maximum rate
$110=4000.000
; 111 - Y-axis maximum rate
$111=4000.000
; 112 - Z-axis maximum rate
$112=4000.000
; 120 - X-axis acceleration
$120=400.000
; 121 - Y-axis acceleration
$121=400.000
; 122 - Z-axis acceleration
$122=400.000
; 130 - X-axis maximum travel
$130=342.000
; 131 - Y-axis maximum travel
$131=244.000
; 132 - Z-axis maximum travel
$132=108.000

`ok
G1X123.909Y5.755Z-9.775
?<Run|MPos:-167.800,-191.133,-76.027|Bf:0,128|FS:1000,12000>
ok
G1X124.415Y6.260Z-9.861
ok
G1X125.004Y6.850Z-10.040
?<Run|MPos:-169.055,-192.385,-75.928|Bf:0,128|FS:1000,12000>
ok
G1X125.510Y7.356Z-10.245
ok
G1X126.015Y7.861Z-10.496
!?<Hold:1|MPos:-170.338,-193.668,-76.465|Bf:0,128|FS:624,12000>
?<Hold:1|MPos:-170.903,-194.233,-76.848|Bf:0,128|FS:0,12000>
?<Hold:0|MPos:-170.903,-194.233,-76.848|Bf:0,128|FS:0,12000>
?<Hold:0|MPos:-170.903,-194.233,-76.848|Bf:0,128|FS:0,12000>
?<Hold:0|MPos:-170.903,-194.233,-76.848|Bf:0,128|FS:0,12000>
?<Hold:0|MPos:-170.903,-194.233,-76.848|Bf:0,128|FS:0,12000>
?<Hold:0|MPos:-170.903,-194.233,-76.848|Bf:0,128|FS:0,12000>
?<Hold:0|MPos:-170.903,-194.233,-76.848|Bf:0,128|FS:0,12000>
?<Hold:0|MPos:-170.903,-194.233,-76.848|Bf:0,128|FS:0,12000>
?<Hold:0|MPos:-170.903,-194.233,-76.848|Bf:0,128|FS:0,12000>
?<Hold:0|MPos:-170.903,-194.233,-76.848|Bf:0,128|FS:0,12000>
?<Hold:0|MPos:-170.903,-194.233,-76.848|Bf:0,128|FS:0,12000>
?<Hold:0|MPos:-170.903,-194.233,-76.848|Bf:0,128|FS:0,12000>
?<Hold:0|MPos:-170.903,-194.233,-76.848|Bf:0,128|FS:0,12000>
?<Hold:0|MPos:-170.903,-194.233,-76.848|Bf:0,128|FS:0,12000>
?<Hold:0|MPos:-170.903,-194.233,-76.848|Bf:0,128|FS:0,12000|Ov:100,100,100|A:S>
?<Hold:0|MPos:-170.903,-194.233,-76.848|Bf:0,128|FS:0,12000|WCO:-292.558,-197.550,-66.000>
?<Hold:0|MPos:-170.903,-194.233,-76.848|Bf:0,128|FS:0,12000>
?<Hold:0|MPos:-170.903,-194.233,-76.848|Bf:0,128|FS:0,12000>
.ok
?
Grbl 1.1ha ['$' for help]
<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0|WCO:-292.558,-197.550,-66.000>
‡‡?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0|Ov:100,100,100>
?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0>
?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0>
?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0>
?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0>
?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0>
?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0>
?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0>
?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0>
?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0|WCO:-292.558,-197.550,-66.000>
?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0|Ov:100,100,100>
?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0>
?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0>
?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0>
?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0>
?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0>
?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0>
?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0>
?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0>
<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0|WCO:-292.558,-197.550,-66.000>
?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0|Ov:100,100,100>
**?<Idle|MPos:-170.903,-194.233,-76.848|Bf:15,128|FS:0,0>
T1
ok
G17
ok
G21
ok
G90
ok
G0Z20.000
ok
G0X0.000Y0.000
ok
S12000M3
?<Run|MPos:-170.903,-194.233,-76.530|Bf:13,128|FS:1920,0>
$I
$I
$I
ok
[VER:1.1ha.20201101:]
[OPT:V,15,128]
ok
[VER:1.1ha.20201101:]
[OPT:V,15,128]
ok
[VER:1.1ha.20201101:]
[OPT:V,15,128]

ok
$I
G4P1
G0X123.341Y4.084Z5.000
[VER:1.1ha.20201101:]
[OPT:V,15,128]
ok
G0Z2.000
G1Z-11.385F1000.0
G1X123.635Y4.377Z-11.260
$+
$G
$#
ok
ok
ok
ok
G1X124.181Y4.923Z-11.123
G1X124.727Y5.469Z-11.104
ok
error:2
[GC:G1 G54 G17 G21 G1X124.894Y5.645Z-11.123
G90 G94 M3 M9 T0 F1000 S12000]
G1X125.246Y6.078Z-11.048
G1X125.509Y6.406Z-11.055
ok
[G54:-292.558,-197.550,-66.0G1X125.615Y6.541Z-11.051
00]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]
ok
ok
ok

ok
ok
ok
G1X125.191Y6.118Z-10.871
G1X124.641Y5.567Z-10.778
G1X124.090Y5.017Z-10.813
error:2
ok
ok
$G
[GC:G1 G54 G17 G21 G90 G94 M3 M9 T0 F1000 S12000]
ok**
‡?<Run|MPos:-279.635,-197.123,-47.573|Bf:4,128|FS:4032,12000|Ov:100,100,100|A:S>
?<Run|MPos:-271.935,-196.868,-48.508|Bf:4,128|FS:4032,12000>
?<Run|MPos:-264.922,-196.635,-49.360|Bf:4,128|FS:4032,12000>
?<Run|MPos:-257.800,-196.400,-50.228|Bf:4,128|FS:4032,12000>
?<Run|MPos:-250.215,-196.148,-51.150|Bf:4,128|FS:4032,12000>
?<Run|MPos:-243.163,-195.915,-52.008|Bf:4,128|FS:4032,12000>
?<Run|MPos:-235.710,-195.668,-52.913|Bf:4,128|FS:4032,12000|WCO:-292.558,-197.550,-66.000>
?<Run|MPos:-228.665,-195.435,-53.770|Bf:4,128|FS:4032,12000>
?<Run|MPos:-221.088,-195.183,-54.693|Bf:4,128|FS:4032,12000>
?<Run|MPos:-213.975,-194.948,-55.558|Bf:4,128|FS:4032,12000>
?<Run|MPos:-206.928,-194.715,-56.415|Bf:4,128|FS:4032,12000>
?<Run|MPos:-199.488,-194.468,-57.318|Bf:4,128|FS:4032,12000>
?<Run|MPos:-192.435,-194.235,-58.178|Bf:4,128|FS:4032,12000>
?<Run|MPos:-184.850,-193.983,-59.100|Bf:4,128|FS:4032,12000>
?<Run|MPos:-177.733,-193.748,-59.965|Bf:4,128|FS:4032,12000>
?<Run|MPos:-171.430,-193.538,-60.730|Bf:4,128|FS:1380,12000>
?<Run|MPos:-169.218,-193.465,-61.123|Bf:5,128|FS:1679,12000>
?<Run|MPos:-169.218,-193.465,-63.930|Bf:6,128|FS:1000,12000>
?<Run|MPos:-169.218,-193.465,-65.720|Bf:6,128|FS:1000,12000>
?<Run|MPos:-169.218,-193.465,-67.505|Bf:6,128|FS:1000,12000>
?<Run|MPos:-169.218,-193.465,-69.365|Bf:6,128|FS:1000,12000|Ov:100,100,100|A:S>
?<Run|MPos:-169.218,-193.465,-71.130|Bf:6,128|FS:1000,12000>
?<Run|MPos:-169.218,-193.465,-72.998|Bf:6,128|FS:1000,12000>
?<Run|MPos:-169.218,-193.465,-74.777|Bf:6,128|FS:1000,12000>
?<Run|MPos:-169.218,-193.465,-76.578|Bf:6,128|FS:588,12000>
?<Run|MPos:-168.773,-193.023,-77.223|Bf:9,128|FS:1000,12000>
?<Run|MPos:-167.488,-191.688,-77.085|Bf:12,128|FS:664,12000>
?<Run|MPos:-167.470,-191.553,-76.920|Bf:14,128|FS:1000,12000>
?<Run|MPos:-168.468,-192.533,-76.813|Bf:15,128|FS:0,12000>
?<Idle|MPos:-168.468,-192.533,-76.813|Bf:15,128|FS:0,12000>
?<Idle|MPos:-168.468,-192.533,-76.813|Bf:15,128|FS:0,12000>
?<Idle|MPos:-168.468,-192.533,-76.813|Bf:15,128|FS:0,12000>
?<Idle|MPos:-168.468,-192.533,-76.813|Bf:15,128|FS:0,12000>
?<Idle|MPos:-168.468,-192.533,-76.813|Bf:15,128|FS:0,12000>
`

terjeio, 2021-11-23:
Another build date issue... `$+` is an extension used by grblHAL and AFAIK GRLB_ESP32 to get all settings, not only those defined by legacy Grbl.
 
Luckily I could fix this and I have uploaded a [new edge build](http://www.io-engineering.com/downloads) with it.

The latest build dates for the official versions are:

gnea/grbl:
`#define GRBL_VERSION_BUILD "20190830"`
gnea/grbl-Mega:
`#define GRBL_VERSION_BUILD "20180813.Mega"`
gnea/grbl-LPC:
`#define GRBL_VERSION_BUILD "20170802"`

Any dates later than these are from ports or non-official builds.

I have a hard time determining which controller is running for build dates > 20200716 so I guess I have to put up a warning about that. Or perhaps a startup option to force Grbl 1.1 mode?
BTW Grbl versions earlier than this is not, and will not be, supported.

terjeio, 2021-11-23:
I have decided to add a [startup option](https://github.com/terjeio/ioSender/wiki/Setup-and-configuration) on the command line to force legacy mode, new edge version uploaded.

calabr, 2021-11-28:
Checked with Sender 2.0.36p11
 Issue solved.
 I think it's not very relyable to use build date. 
 In case of any bug-fix will be released platform detection will breack.
  In my opinion, right wey to detect platform based on version ( GRBL1.1 - always legacy) 
 grblHAL - isn't GRBL any more and should repart different name.
 If you want to keep compatebility of HAL with legacy sender - probable better to add configuration switch in the controller and switch legacy version reporting, switch-off some HAL-specific behaviour etc.
 
For ioSender I suggests to add version override option to the config file, like "Auto, grbl1.x, grblHAL..."
In future you can add GUI support for it.

closed 2021-11-28 by calabr

terjeio, 2021-11-28:
> In my opinion, right wey to detect platform based on version ( GRBL1.1 - always legacy)

Not possible since most of the controllers grblHAL supports is not, and some can not be, reset upon connecting. So no welcome message. Catch 22.