terjeio/ioSender #163: Probe setup issues

calabr, 2021-11-29:
Probe setup don't allow to to select G38.2 mode. As result, if first try failed, probe going additional 2mm with slow speed, instead of stop with alarm.
Stop button don't interrupt current test run, as result, if you see simething going wrong, you should wait your probe crash or run to press E-stop.
Today I'm broke mine during the test :(,
I cannot reproduce scenario, but seems like this:
  during Height map test probe found the hole in sacrificial table and didn't trigger, 
stop button pressed without result
Then I'm pressed hardware Stop or Feed Hold and switched to GRBL tab to reset controller.
 At this moment machine started to move horisintally with probe tip in the hole.

It is no any clear readable indicator of test running state and finished/failed resone isn't raedable (small note in the status-bar)
DRO screen and main controlls should alwais be present on the screen. 
Probe progress neeed clear indication of current state.
Errror indication should be **BIG red** and don't disapear, overwritten by regular events.
I'm suggesting probe algorithm like this:
 ```
x,y,z = mx,my,mz    	# remember position
 g91 G38.2 F<fast_feed>X<dx>Y<dy>Z<dz>
 g91 G38.4 F<fast_feed>X[x-mx]Y[y-my]Z[z-mz] # move back - max. position start point
x2,y2,z2 = mx,my,mz  # remember start point for slow speed move
 g91  G38.2 F<slow_feed> X[x2-mx]Y[y2-my]Z[z2-mz]
 Collect probe position
G53G0 X[mx] Y[my] Z[mz]         # Return to start point
```
Also G38,2/G38.4 and G38.3/G38.5 should be selectable for NC probe 
Probe result point should be indicated somewhere on the screen.
if user don't set offset checkbox before test, it's should be button to apply offset after probe completted.
Preview checkbox cause missunderstanding - I'm pushed Start button, but nothing started - I'm trying to push it again.
 Suggesting to add Preview button, instead of checkbox and let Start button to start probbing.
Height map page : "Set from program limits" button resets grid size to 5x5
     3D view window show nothing - I'm apologise it should be work-pice miniature and probe grid over it.

terjeio, 2021-11-29:
> Probe setup don't allow to to select G38.2 mode. As result, if first try failed, probe going additional 2mm with slow speed, instead of stop with alarm.

Is `#define MESSAGE_PROBE_COORDINATES` commented out in your configuration? I use that that for handling program flow.
I guess I should switch to G38.2 for legacy Grbl as there is no way to detect this config.

> Stop button don't interrupt current test run, as result, if you see simething going wrong, you should wait your probe crash or run to press E-stop.
Today I'm broke mine during the test :(,

Sorry to hear that, and is due to a coding error on my side. For legacy Grbl I have to send a reset command as there is no other way to terminate a motion. grblHAL has a new real-time command for stopping motion without triggering a soft reset and this was incorrectly sent in legacy Grbl mode. Fixed in the next build. Note that I am not able to test this since I do not have a legacy GRBL controlled machine.

> Also G38,2/G38.4 and G38.3/G38.5 should be selectable for NC probe

G38.4/G38.5 has nothing to do with NC probe, it is for probing away from the workpiece when it is already triggered. Perhaps a new probing tab should be added for that? Use the `$6` setting to invert the probe signal for NC probes.

> if user don't set offset checkbox before test, it's should be button to apply offset after probe completted.

Which offset? From the first probing result? You can use the _Tool length offset_ tab to set the offset anywhere you like.

> Preview checkbox cause missunderstanding - I'm pushed Start button, but nothing started - I'm trying to push it again.
Suggesting to add Preview button, instead of checkbox and let Start button to start probbing.

I'll look into that, should not be to hard to change.

> Height map page : "Set from program limits" button resets grid size to 5x5

I'll fix this.

> 3D view window show nothing - I'm apologise it should be work-pice miniature and probe grid over it.

This is already on my todo list.

calabr, 2021-11-30:
> Is `#define MESSAGE_PROBE_COORDINATES` commented out in your configuration? I use that that for handling program flow. I guess I should switch to G38.2 for legacy Grbl as there is no way to detect this config.

No it's not commented, like in default setup.
I think G38.2 or G38.3 should be selectable in probe setup frame.

> Sorry to hear that, and is due to a coding error on my side. For legacy Grbl I have to send a reset command as there is no other way to terminate a motion. grblHAL has a new real-time command for [...]
 
Never mind.
If you need to test some particular cases - don't hesitate to ask.

> > Also G38,2/G38.4 and G38.3/G38.5 should be selectable for NC probe
> 
> G38.4/G38.5 has nothing to do with NC probe, it is for probing away from the workpiece when it is already triggered. Perhaps a new probing tab should be added for that? Use the `$6` setting to invert the probe signal for NC probes.

$6 is the global setup.
Depending on current needs NC or NO probe can be used.
Not taking in account standart description - 
G38.2 move until signal rised UP, G38.4 - move until signal DOWN and corresponding state check to start probbing. 
For NO G38.2 to find work peece, G38.4 to move out of it
For NC - just inversion G38.4 to find work peece, G38.2 to move out of it
 
> > if user don't set offset checkbox before test, it's should be button to apply offset after probe completted.
> 
> Which offset? From the first probing result? You can use the _Tool length offset_ tab to set the offset anywhere you like.

From last probe (not for height map)
Also last probe result cannot be seen in GUI.
Offset page don't allow to use last probbing result to set offset, also touch-plate and probe size not involved.

During switching beatween offset and probbing page exeption rised
[image]

After doing Tool Length probe and switch to any tab and back new button appear on the screen.
 "Clear Tool length Offset" it's where not appear without tab switch
[image]

terjeio, 2021-11-30:
> I think G38.2 or G38.3 should be selectable in probe setup frame.

I could make that an option for grblHAL but not for legacy Grbl as there is no way to detect how it is configured. So for legacy Grbl I have changed it to G38.2 only.

> G38.2 move until signal rised UP, G38.4 - move until signal DOWN and corresponding state check to start probbing.

G38.2 is probe towards workpiece, G38.4 is away from it, from LinuxCNC specs:

>• G38.2 - probe toward workpiece, stop on contact, signal error if failure
• G38.3 - probe toward workpiece, stop on contact
• G38.4 - probe away from workpiece, stop on loss of contact, signal error if failure
• G38.5 - probe away from workpiece, stop on loss of contact

Using G38.2/G38.4 to switch between NC/NO probes is IMO not a compliant use.

> For NO G38.2 to find work peece, G38.4 to move out of it
For NC - just inversion G38.4 to find work peece, G38.2 to move out of it

I use rapid motion to retract after probe is triggered, using G38.4/G38.5 makes no sense?

> From last probe (not for height map)

Then I do not understand what you mean. The other probing screens (other than rotation) allows setting the offset from either the last point probed or from the calculated point.

> Also last probe result cannot be seen in GUI.

If you need them all probing results are available from the console. This can be made floating from the _File_ menu.

>Offset page don't allow to use last probbing result to set offset, also touch-plate and probe size not involved.

You can set whatever values you want on the offset page, not sure what your use-case is. BTW you can change the offset by clicking in the DRO value fields, edit and commit by pressing the enter key.

> During switching beatween offset and probbing page exeption rised

Does not happen for me, I have fixed this in the latest version without knowing?

> "Clear Tool length Offset" it's where not appear without tab switch

I'll have to check this somehow - I have tried to code around legacy Grbl limitations for this.

I have uploaded a new [edge version](http://www.io-engineering.com/downloads) for you to try with some of the changes you want, G38.2 for probing and changing the _Stop_ button label according to the current controller status. The _Stop_ button in the probing tabs now sends a soft reset for legacy controllers.

calabr, 2021-12-01:
Checked with sender 2.0.36.p13
 With G38.2 and reset on stop - now it's more reliable.

> I use rapid motion to retract after probe is triggered, using G38.4/G38.5 makes no sense?

Rapid speed will never be reached with resonable retract becaus of acceleration.
It's much more comfortable to use - no need to think how much to retract for second touch. Also it can be faster for scanning task.
 The fail scenario with manual retract: set retract for 5mm, max travel 10mm. real distance to workpiece 0.8mm
 most lexible to allow user select what he need.
Also latch travel distance 1.5xRetract - is to far. Better to restict it with previous search point.

> Then I do not understand what you mean.

I think I know where is miss-understending - 
 Tool length page implicitly set TLO (G43.1) if offset checkbox not selected. - When I look at this page - I think it will only find point and allow me to decide what to do.
 It's better to set  explicit selector TLO/Offset

looks like I found enother GRBL bug - setting Offset by G10L20 and immediate move throw an error

```
?<Run|MPos:-217.575,-123.678,-80.005|Bf:14,128|FS:25,0>
[PRB:-217.575,-123.678,-80.063:1]
ok
G10L20P1Z2.5
G53G0Z-75.35
ok
error:2
$G
[GC:G38.3 G54 G17 G21 G91 G94 M5 M9 M56 T0 F25 S0]
ok
G90

```
If I'm doing the same manually - all is ok.
Command sequence on edge-finder page work without problem
```
?<Run|MPos:-291.630,-148.480,-37.415|Bf:14,128|FS:0,0|Pn:P>
[PRB:-291.630,-148.480,-37.400:1]
ok
G53G1F200Z-37.015
ok
G53G1F200Y-148.48
ok
?<Run|MPos:-291.630,-148.480,-37.403|Bf:14,128|FS:200,0|Pn:P>
G53G1F200X-291.63
ok
?<Run|MPos:-291.630,-148.480,-37.345|Bf:14,128|FS:200,0|Pn:P>
G10L2P1Z40.865
G91
‡ok
ok
?<Idle|MPos:-291.630,-148.480,-37.015|Bf:15,128|FS:0,0|WCO:-291.630,-148.480,-39.900>
```

terjeio, 2021-12-01:
> It's much more comfortable to use - no need to think how much to retract for second touch. Also it can be faster for scanning task.
The fail scenario with manual retract: set retract for 5mm, max travel 10mm. real distance to workpiece 0.8mm
most lexible to allow user select what he need.

I still do not get it - you mean use G38.4 instead of retracting and then skip the second slower pass?

> Also latch travel distance 1.5xRetract - is to far.

Too far for what reason? Still a risk of triggering soft limits?

> Better to restict it with previous search point.

It should at least be a little bit more or the probe may not trigger...

> When I look at this page - I think it will only find point and allow me to decide what to do.

The tab is named _Tool length offset_ si IMO it is pretty clear what it is intended for. You are better served by writing a macro? FYI I have added [parameter](http://www.linuxcnc.org/docs/html/gcode/overview.html#_parameters) and [expression](http://linuxcnc.org/docs/2.5/html/gcode/overview.html#sec:Expressions) support in the edge version. Parameters 5061 - 5070 is probe data. Note that I have not yet tested this extensively. 

> looks like I found enother GRBL bug - setting Offset by G10L20 and immediate move throw an error

No, it is likely a Grbl feature. I'll have to write a workaround for that too. G10L20 immediately writes data to EEPROM, this disables all interrupts leading to loss of data. grblHAL buffers writes to RAM...

calabr, 2021-12-01:
> I still do not get it - you mean use G38.4 instead of retracting and then skip the second slower pass?
 
G38.4 instead of retract with limit of starting point, than second slower pass
  Reading data on retract pass also can be helpful for some bouncing probes.

> > Also latch travel distance 1.5xRetract - is to far.
> 
> Too far for what reason? Still a risk of triggering soft limits?

for soft limits, for crashing probe - very relevant for probbing PCB with graver.

> > Better to restict it with previous search point.
> 
> It should at least be a little bit more or the probe may not trigger...

 Yes you right - It can be configurable overshoot limit.
 
> > When I look at this page - I think it will only find point and allow me to decide what to do.
> 
> The tab is named _Tool length offset_ si IMO it is pretty clear what it is intended for. 
  
For me it where become not clear it will set this offset, or just collect data.
> You are better served by writing a macro? FYI I have added [parameter](http://www.linuxcnc.org/docs/html/gcode/overview.html#_parameters) and [...]
 
Yes! it's greate thing! I'll try it.

> No, it is likely a Grbl feature. I'll have to write a workaround for that too. G10L20 immediately writes data to EEPROM, this disables all interrupts leading to loss of data. grblHAL buffers writes to RAM...

Now it's also working fine.
Thanks alot for this improvements!

closed 2022-05-17 by calabr