terjeio/ioSender #171: Console window frize the program

calabr, 2021-12-05:
ioSender 2.0.36p15
Controller GRBL 1.1h Atmega328

Started to send code.
After switching Console Verbose output programm may stall in ~10 sec. or after switch from Program tab to Console tab 
It's continue to send "?" requests, but stopped to send GCode and do not react on any button. 
App window cannot be closed. Only killed from task-manager.
1 CPU core become 100% loaded.
 Looks like it's only happen with "-locale ru-RU" switch. I cannot get such issue with English or de-DE.

terjeio, 2021-12-08:
I am not able to replicate this. Does the gcode contain a lot of short movements?

FYI I sometime get hangs while debugging controller code due to the controller USB stack dying. Usually I can then terminate ioSender by unplugging the USB cable.

calabr, 2021-12-14:
Yes it's 25K-100K strings of short moves. 
I get the same issue with GRBL in CHECK mode and with free running ( motors switched off,  feed override 200% for stress-test.) 
  My setup use CH340 USB-UART convertor. 
 Serial port monitor show the successfull "?"  from  sender and normal reply from grbl. 
I think it's something happens with ok response  parser. 
 Locale type influence - probable some issue with string length.  My translation is usually longer than English.

terjeio, 2021-12-14:
I am still not able to replicate this after sending a couple of million lines in Check mode to an Arduino Mega. I have tested with both the Russian and German translations. Can you test with another computer and/or USB cable? Or post a file that fails for me to test with?

terjeio, 2021-12-17:
By chance I saw a similar (or the same?) issue when starting a second run of a g-code file that was not properly ended (by a command such as M30). Can you check with the [latest Edge version](https://github.com/terjeio/ioSender/discussions/176#discussioncomment-1838643) to see if the issue is resolved for you?

closed 2022-09-27 by terjeio