terjeio/ioSender #172: ioSender assumes Grbl is in mm mode (G21)

phil-barrett, 2021-12-07:
ioSender 2.0.35

If a Gcode program uses a G20 (Imperial Mode), once it completes, all jogs are 25.4 farther.  I think speed is also affected.

Work around is to issue a G21 via MDI after running a job that uses imperial.

terjeio, 2021-12-07:
If I have done it correctly this is fixed in 2.0.36, I need to retest after moving some code for the XL version.

phil-barrett, 2021-12-07:
Great!  FWIW, Edge 2.0.36p15 has the issue.

terjeio, 2021-12-08:
> Edge 2.0.36p15 has the issue.

Works for me, but then I have _Parser_ state enabled in _Status report options_. I guess I have to request a parser state report on the end of a job. Note that I am not keen to start parsing the the gcode as it is sent.

closed 2023-04-29 by terjeio