terjeio/ioSender #175: Setting to disable perspective in 3D view

LaurentLM, 2021-12-09:
Hi,
A small detail, but I was wondering if the next version could have a setting to disable the perspective in the 3D view.

This would be useful when setting a job. At the moment, the parallax makes it tricky to see what toolpaths are directly under the tool. 
I hope this makes sense.
Many thanks!

terjeio, 2021-12-13:
New [edge version](http://www.io-engineering.com/downloads) uploaded with a combo box for selecting view mode.

[image]

calabr, 2021-12-18:
This feature is greate!

In ioSender 2.0.36p21
 After any move in plane view XY, XZ, YZ switching to different plain show incorrect position Like cutter located over axis. Any move command cause cutter mark to jump correct position.
Switching XZ <-> YZ didn't rotate view. XY <-> XZ and XY<->YZ - rotates normally
switching XY to XZ with "Show axis" option cause X axis gray cone cover all window.

in plane view job bounding box and machine boundary shown with perspective, which is confusing.
With close look in XY plane, red cutter circle cover trace marks and not allow to precise locate it. It's better to switch it not filled circle with center cross

terjeio, 2021-12-18:
New version uploaded, please add any further comments to the [edge version discussion](https://github.com/terjeio/ioSender/discussions/176#discussioncomment-1841826).

closed 2021-12-18 by terjeio

LaurentLM, 2021-12-19:
I have'nt had a chance to test that properly, but it looks good. Thanks for adding this feature.