terjeio/ioSender #177: Keyboard jogging does not seem to use grblHAL settings ($50-55)

phil-barrett, 2021-12-17:
This is with the  T4.1 port. 

It looks like step jogging (<ctrl>+cursor) is always .1 mm no matter what $53 is set to.  I tried setting slow jog rate and distance to crazy number ($51=6000, $54=5000) and still get fairly short/slow movement.  Perhaps I am doing something wrong?

Bert911, 2022-01-29:
I'm not seeing any of the setting ($50-55) with the new Iosender 2.0.36 but like the new XL interface very much.

I'm using also the T4.1 with Ethernet and using Phil's prebuilt version from the 03.01.2022.
It's no big trouble but it would be nice to have jogging above 500 mm/min again.

Manually changing $51 any setting ($50-55) result in an error:

error:53 - Setting is not available, possibly due to limited driver support.

terjeio, 2022-01-29:
Settings `$50-55` is only available if the keypad plugin is compiled in, if not keyboard jogging feedrates are set in the _Settings: App_ tab.
New is that continuous keyboard jogging has been disabled due to user requests - by default keyboard jogging uses parameters from the UI jog panel, in the same way that most other sender does. This can be overridden by creating a shortcut to the executable and [adding a command line parameter](https://github.com/terjeio/ioSender/wiki/Setup-and-configuration) (`-jogmode`) to select which type of jogging to use.

Bert911, 2022-01-29:
I'm feeling like you're trolling me  :grin:

The settings from the UI Jogging differs from my keyboard jogging.
Keyboard Jogging uses 200 mm/min, 500 with "shift" and 100 with "ctrl".

In the _Settings: App_ tab I don't have these settings any more, but I know this settings from the older versions of Iosender.

What am I missing?

[image]

terjeio, 2022-01-29:
> The settings from the UI Jogging differs from my keyboard jogging.
Keyboard Jogging uses 200 mm/min, 500 with "shift" and 100 with "ctrl".

A bug?, I'll have to check that.

With these shortcut parameters:

[image]

I get:

[image]

And with `-jogmode 2` I get both the _UI jogging_ and _Keyboard jogging_ panels.

Bert911, 2022-01-29:
I may have missed the "1" in -jogmode 1.
With `-jogmode 2` I get the known settings, but they don't work :neutral_face:

terjeio, 2022-02-05:
Can you try the [latest edge version](http://www.io-engineering.com/downloads)? There was a bit of code missing in the XL build.

Bert911, 2022-02-05:
Works like a charm with the new version.

Thank you very much!

closed 2023-01-29 by terjeio

prcdslnc13, 2023-07-18:
I am getting the same issue with the latest edge builds and 2.0.42. No keyboard Jogging pane with -jogmode 2 in the parameters. 
[image]

[image]

It's not working with the XL or normal version.

terjeio, 2023-07-19:
@prcdslnc13 Do you have the keypad plugin active in the controller? If so the keyboard jog settings are found in the _Settings: Grbl_ tab and inherited from there.

prcdslnc13, 2023-07-19:
Oh interesting. I do have keypad active. Ill try that. Thank you!

prcdslnc13, 2023-07-24:
Ok what I found was the Keypad settings were not respected but I was able to change the speed in the app.config file here 
[image]