terjeio/ioSender #180: TMC tab runaway error

Cncninja, 2021-12-18:
I'm trying to set up my TMC 5160 and I can see them and change prams but when I try to use sensorless homing and I get an error screen pop up that runs away making 100's of windows. the only stop is to reboot. Also, no matter what settings I use the sensorless homing does not work it just jams the axis and keeps trying to go till i get a error 9. I have sensorless homing enabled and homing, and the diag pin is connected (5160 it was not installed) its a SKR 1.4 turbo board LPC1769
[image]

terjeio, 2021-12-18:
Try the latest [edge version](https://github.com/terjeio/ioSender/discussions/176#discussioncomment-1841826), that works for me.

Cncninja, 2021-12-19:
yes I'm running the most current version edge XL

terjeio, 2021-12-19:
36p22?

terjeio, 2021-12-19:
The runaway error messages popped up for me when testing Trinamic drivers with ESP32, but not when testing STM32F4xx. With the latest commit just uploaded they disappeared for the ESP32 driver, perhaps this wil solve it for the LPC176x driver as well?

Sensorless homing can be tricky to tune, I suggest you start with a free running motor an load the motor by hand to see how the stallguard response changes. Use G1 commands with an appropriate feed rate when doing this.
IMO sensorless homing is not well suited for builds that has to be optimized by end users, I suggest you take a look at the Trinamic driver documentation for information about how to do it. Note that there are additional tuning parameters in [trinamic.h](https://github.com/grblHAL/Plugins_motor/blob/e5d64624161032d34c8ea72ffd31fbc50632782f/trinamic.h#L70-L71) that is not made available via settings.

Also, there is a "bug" in ioSender that shows up when _Reset_ is pressed in the _Trinamic tuning_ tab, the reset causes the driver to go back to normal operation mode - not StallGuard mode which is required for tuning. I'll trap the reset in the next edge build and switch the selected driver back to StallGuard mode, for now you have to leave the tab and reopen it.

Cncninja, 2021-12-22:
I was on 21, just updated, and seems ok and sensorless homing started to work, I was able to get it dialed in.

Cncninja, 2021-12-26:
35p22 fixed the issue

closed 2021-12-26 by Cncninja