terjeio/ioSender #182: Settings:GRBL and Feedrate GUI not showing

BradGilmore, 2022-01-02:
I recently updated to 2.0.35 and now the $$ settings no longer give a description in Settings: GRBL.

Also, when playing with the feed rate and spindle speed over ride, the readout doesn't show anything was changed but the machine does react.

Lastly, is it possible to add a simple return to zero button? It seems that no matter what code I issue, or macro I create to return to zero, it's off in a random direction on the XY plane but Z seems accurate.

Thanks for all you do and the improvements so far have made a big difference.

phil-barrett, 2022-01-02:
Are you using a Teensy 4.1?  If so, you probably need to update to a newer build of grblHAL.  If you are using the T41U5XBB, there are prebuilt binaries available [here](https://github.com/phil-barrett/grblHAL-teensy-4.x/blob/master/hex_files/grblHAL_T41.zip).

BradGilmore, 2022-01-02:
Yes I am using the Teensy 4.1 with the T41U5XBB v2.09 board. Will the newer version of grblHAL solve these issues?

phil-barrett, 2022-01-02:
Yes, the older prebuilt binaries were made before the grblHAL update to support ioSender Grbl Setting section.

I don't know if it will fix the other issues you are having but I highly recommend updating grblHAL. It fixes a Teensyduino bug that would lead to random problems.

BradGilmore, 2022-01-02:
I'll give that a try and report back. Thank you.

BradGilmore, 2022-01-02:
Is there a specific way to update or do I drag and drop the files?

phil-barrett, 2022-01-03:
Yeah, use teensy.exe (from pjrc.com). Follow the instructions [here](https://github.com/phil-barrett/grblHAL-teensy-4.x/blob/master/prebuilt.md).

djuuru, 2022-01-08:
Had very similar issues and updating to the latest prebuilt binary fixed the $$ settings descriptions for me.

But spindle / feed rate override readout issue still remains though.
Also, trying to move the machine by updating the DRO fields does nothing.. Don't know if it's connected..

BradGilmore, 2022-01-08:
I can confirm the same as djuuru said so far. I am still in the process of re configuring my settings to get the machine to move.

BradGilmore, 2022-01-08:
Ok, machine is up and moving. I have updated the board and all is well so far. I did notice while testing that the spindle override DOES work and gives proper RPM updates as well. I have not tested the feed override yet though.

terjeio, 2022-01-08:
> But spindle / feed rate override readout issue still remains though.

I cannot replicate this with build 2.0.35 nor the latest [edge build](http://www.io-engineering.com/downloads). Be aware that feed override will not take the feed rate beyond the configured max rate.

> Also, trying to move the machine by updating the DRO fields does nothing.. Don't know if it's connected..

Updating a DRO field is for [setting the current position](https://github.com/terjeio/ioSender/wiki/Usage-tips#setting-current-position) to whatever you want it to be, not for motion. And \<Return\> or \<Enter\> has to be pressed to set it. Use the _MDI_ input field to type and send commands.

djuuru, 2022-01-09:
> Updating a DRO field is for [setting the current position](https://github.com/terjeio/ioSender/wiki/Usage-tips#setting-current-position) to whatever you want it to be, not for motion. And <Return> or <Enter> has to be pressed to set it. Use the _MDI_ input field to type and send commands.

Ok, my bad. I misunderstood that. 
Although "zero all" (or any of the separate axis 0 buttons) does not update the DRO values either. And changing the DRO field and pressing <Enter> just goes back to whatever the value was before editing the field.

If you can't replicate it - don't worry about it. I'll try the edge build and maybe try to recompile latest grblHAL at some point. Not a big problem, just thought I'd let you know. 

Awesome work BTW, really appreciate the software!

terjeio, 2022-01-09:
> and maybe try to recompile latest grblHAL at some point

Which version do you have?

djuuru, 2022-03-26:
Finally had the time to recompile the latest grblHAL from source and it fixed my issues. Even got probing to work. 

Recovering grbl settings from the backup file gave me a bunch of errors.. so I just reconfigured stuff I needed from scratch and now works like a dream! So probably had something configured wrong.. 

So as far as I'm concerned - issue solved.

closed 2023-04-29 by terjeio