terjeio/ioSender #184: Z Height Probe

BradGilmore, 2022-01-08:
I'm trying to do a simple Z height probe in the Probing:Tool Length Offset tab. When I complete the probe, it doesn't seem to take into account the probe height when setting zero. I may be doing this wrong as I'm still learning Gcode.

The probe touch plate height is 19.15mm. Probing distance is approximately 3mm of travel to the touch plate. 
After probing and retract, Z is set to -13.965mm. Shouldn't the Z height be at least +19.15 (probe plate height) plus whatever the retract distance is?

Am I doing something wrong? Also, is there an option to change the retract distance after probing?

terjeio, 2022-01-08:
> I'm trying to do a simple Z height probe in the Probing:Tool Length Offset tab. 

For simple probing it is perhaps better to use _Edge finder, external_ and click in the center of the image to select Z probing.

> After probing and retract, Z is set to -13.965mm. Shouldn't the Z height be at least +19.15 (probe plate height) plus whatever the retract distance is?

That depends on what is selected/set in the UI, if  only _Add offset_ is checked the touch off position is set to _Touch plate_ heigth plus _Workpiece Height_. If the touch plate is placed on the top of the workpiece Z0 will be at the top of the workpiece if _Workpiece Height_ is set to 0, and at the bottom of it when set to the actual workpiece height.

If only _Establish reference offset_ was checked this does not change any offset and the probed position is used to calculate and set tool offset after subsequent touch off when changing tool with the TxM6 command.

> Also, is there an option to change the retract distance after probing?

No, the tooltip is always moved back to the real world Z position it had before probing started.

BradGilmore, 2022-01-08:
I must be missing something. The "Add offset" is only in the Tool Length Offset tab that I can see.

I did find the error which was on my part. After updating Teensy, I missed setting the steps/mm for the Z axis..

I tried the probing again and then sent Z0 and now the tool tip moves to the work piece as it should. I wrote a macro during the process that also confirmed everything works properly. Also, using this macro allows me to set the retract height to whatever I need.

G91
G38.2 Z-25 F100
G04 P0.25
G0 Z1
G38.2 Z-5 F25
G04 P0.25
G10 P0 L20 Z19.15
G0Z5
G90

Thank you for taking the time to help!

closed 2022-01-20 by terjeio