terjeio/ioSender #187: Height map calculation issue

calabr, 2022-01-16:
Applying Height map to the program code cause starange move on Z
test code:
```
G21 G17 G90
G0Z20
G0X0Y0
Z1
G1X10F100
Y2
X0
G0Z5
M30
```
resulting after correction:
```
G17G21G90G0Z-4.153
G0Z-3.615
G0Z-3.078
G0Z-2.54
G0Z-2.003
G0Z-1.465
G0Z-0.928
G0Z-0.391
G0Z0.147
G0Z0.684
G0Z1.222
G0Z1.759
G0Z2.297
G0Z2.834
G0Z3.372
G0Z3.909
G0Z4.446
G0Z4.984
G0Z5.521
G0Z6.059
G0Z6.596
G0Z7.134
G0Z7.671
G0Z8.208
G0Z8.746
G0Z9.283
G0Z9.821
G0Z10.358
G0Z10.896
G0Z11.433
G0Z11.97
G0Z12.508
G0Z13.045
G0Z13.583
G0Z14.12
G0Z14.658
G0Z15.195
G0Z15.733
G0Z16.27
G0Z16.807
G0Z17.345
G0Z17.882
G0Z18.401
G0Z18.945
G0Z19.475
G0Z20
G0X0Y0Z20
G0Z15.25
G0Z10.5
G0Z5.75
G0Z1
F100
G1X5Z1.02
G1X10Z0.992
G1Y2Z1.006
G1X5Z1.017
G1X0Z0.998
G0Z4.998
M30
```
  Looks like current machine position influence correction calculation - position was -4.772 at the moment of apply.
as you can see, simple vertical move translated to the series of small move begining from current position.
 If I apply correction, and then set tool 0 - first move will put cutter far below cutting surfece
[test.map.txt](https://github.com/terjeio/ioSender/files/7877787/test.map.txt)
.

terjeio, 2022-01-23:
I believe I have fixed this in 2.0.36p28 edge version. Can you test and confirm?

calabr, 2022-02-12:
Checked with 2.0.37p1
Ok

closed 2022-02-12 by calabr