terjeio/ioSender #188: Nothing happens when iosender.exe is clicked

DavidR8, 2022-01-17:
Running Win10, downloaded the most recent version, unzipped, clicked on iosender.exe. Message that no config file exists do I want to create a new one. I click yes. Nothing further happens.
With Task Manager open clicking on iosender.exe again I see the app in the list then it says suspended and disappears.
I have no idea how to actually get the app running.

calabr, 2022-01-18:
Do you have any com port in the system when starting application?
It's bug in software 
https://github.com/terjeio/ioSender/issues/149#issuecomment-965487685
which can correlate with yours.

DavidR8, 2022-01-18:
@calabr I figured it out. I needed to actually have a suitable board connected.

Guenni75, 2022-04-14:
How can i start without a connected board???????

calabr, 2022-04-14:
> How can i start without a connected board???????

 It's not supported. 
You can try to use simulator, but you should compile it yourself.
[grbl-sim](https://github.com/grblHAL/Simulator)

closed 2023-04-29 by terjeio