terjeio/ioSender #190: Issue 105 (random moves after stopping) still happening

jappiemike, 2022-01-26:
The issue was closed because of a lack of feedback, but I've also noticed this behavior using IOSender 2.0.35 and grblhal from 19/9/21.

terjeio, 2022-01-26:
What kind of moves? Keyboard  jogging? MDI input? Probing? All?
Does it happen often?

Which grblHAL driver?

terjeio, 2022-01-27:
>  Like if you are doing a probing motion and change your mind and press stop.

I saw that happening a short while ago and fixed it in recent [edge versions](http://www.io-engineering.com/downloads). I has never been reported so I decided to delay a fix until 2.0.36 is released, I'll do that ASAP.

jappiemike, 2022-01-27:
I've noticed it right after pressing STOP during a program run and trying to raise the spindle. I'm using a STM411 board.
Cheers
Mike

terjeio, 2022-01-28:
More details please as I am not able to reproduce this with the [2.0.36 release](https://github.com/terjeio/ioSender/releases/tag/2.0.36).

Does it happen every time?
If not how often? 1 out of 10 times...?
Is the unwanted motion starting when you press the page up key or immediately when pressing the _Stop_ button?
What kind of moves? Z only?
Same direction every time?
Since it is when running a loaded program does it only happen with a few or all?

jappiemike, 2022-01-29:
OK I've done a bit of testing, here are the results
cycle start - altS or stop button the spindle runs on X&Y random move Z doesn't appear to move
cycle start - feed hold then altS the spindle runs on X&Y random move Z doesn't appear to move
cycle start - space bar then stop button the spindle runs on X&Y random move Z doesn't appear to move
To stop the spindle M5 doesn't work you must use M2 (program end)
If I set a zero position (x-362 Y-530 Z-28) and then run a program and then stop the program and  use G90 G0 X0 Y0 to return to zero the DRO  shows 3 zeros but the tool is in the wrong place and there is no jogging with either the keyboard or jogging window and the MPOS window co-ordinates are also wrong possibly by the random move amount.
I've also noticed that homing doesn't work afterwards with the Y axis alarming on the rear proximity switch and the X axis stops moving when this happens. The second homing works correctly.
I haven't updated either GRBLHAL or IOsender 
I hope this is as clear as mud and of some help.
Cheers
Mike

terjeio, 2022-01-29:
I was able to replicate the spindle not stopping/M5 not working issue and have fixed that in the latest commit of grblHAL.

I cannot replicate the rest with 2.0.36 (except your homing issue, I suspect that has nothing to do with the sender), can you update and check again on your end?

jappiemike, 2022-01-30:
I've used both 2.0.36 and the XL versions and updated GRBLHAL to 20220129 with driver version 220111 for the 411 board and they both still have the random moves. Thanks for fixing the spindle problem.
Cheers
Mike

terjeio, 2022-01-31:
> ...This happens every time the program is stopped.
cycle start - altS or stop button the spindle runs on, when jogging, X&Y random move Z doesn't appear to move

Please post a g-code program where this happens.

jappiemike, 2022-02-02:
Here is a small program.
[LocatingHoles.txt](https://github.com/terjeio/ioSender/files/7984399/LocatingHoles.txt)

terjeio, 2022-02-04:
I have now tested your program with several drivers, including F41, and and I am still not able to replicate.
Do you get the movements with both UI panel and keyboard jogging?
If you send a MDI command for moving the Z-axis does that trigger the movements as well?

jappiemike, 2022-02-06:
Having done some more testing, if you stop the program mid run and use the MDI panel, using either the Z+ or Z- buttons causes the random movement in X and Y but no Z movement. On the other hand using the keyboard PGUP or PGDN keys causes a XY movement followed by the Z movement. I have the backlash compensation turned on and set to 0.2mm, although the random movement is greater than this, could this be part of the problem?
Cheers
Mike

terjeio, 2022-02-06:
> I have the backlash compensation turned on and set to 0.2mm, although the random movement is greater than this, could this be part of the problem?

It seems so, the position for backlash calculation is not reset on stop. Add these lines [after this line](https://github.com/grblHAL/core/blob/45451a1ce79bebba5ab0bebb473c40fdd083cfac/protocol.c#L521) and retest:
```
#ifdef ENABLE_BACKLASH_COMPENSATION
            mc_backlash_init();
#endif
```

No wonder I could not replicate the issue before...

jappiemike, 2022-02-10:
Hello and sorry for the late reply. It is still not working correctly. After stopping the program if you jog the Z axis it still moves the X&Y first and if you move either X or Y the Z axis still moves as well. Thank you for trying to solve this glitch.
Cheers
Mike

terjeio, 2022-02-10:
By how much do they move?
I you set backlash to 0 do they still move?
Does the motion happen regardless of which direction you jog the Z-axis?

jappiemike, 2022-02-11:
OK, it's definitely the backlash compensation, if I set them to 0 there are no random moves. I did put in the extra lines you asked for before this test. I've also noticed that the shortcut keys for starting, feed hold and stop are no longer working, is there a setting for these I've changed by mistake?
Cheers
Mike
Do you have a patreon where I can throw some money?

terjeio, 2022-02-11:
> I did put in the extra lines you asked for before this test.

Are you sure they were compiled in for the test...?
Anyway, I have changed the grblHAL code a bit more - the backlash code keeps track of the last directon for each axis. This was reset to default on a stop/soft reset and could lead to random movements - but only by the configured backlash distance and when the axis in question was commanded to move. The change, just commited, keeps the last direction over a stop/soft reset to avoid this.

> I've also noticed that the shortcut keys for starting, feed hold and stop are no longer working, is there a setting for these I've changed by mistake?

Odd, there are no settings for those (yet). You may delete _App.config_ to start with a clean configuration.

> Do you have a patreon where I can throw some money?

No, I do not have that.

closed 2022-02-13 by jappiemike