terjeio/ioSender #195: IOSender doesn't resume after semi-auto tool change.

andrewmarles, 2022-01-31:
At the T4M6 line below, IOSender doesn't resume after the tool change is completed, it just stops in idle state on that line:

TLO is set via probing, tool change executes correctly and the setter is probed at G59.3, and the offset is applied correctly.  If I execute the job with T4 already loaded and set, the program runs successfully.

This is with 2.036 on the IMX driver, latest firmware version.  Issue is also present in 2.035

Also worth noting, is that if I run this with  aggressive buffering, not only does the program stop at the tool change, it also turns on the spindle at the start of the tool change procedure - I think this is because with aggressive buffering the S21000 M3 line has been sent and is in the buffer, and the first thing the tool change procedure does is sync the buffer.

`G90 G94
G17
G21

(Adaptive3)
T4 M6
G4 P1.
S21000 M3
G4 P3
G54
M7
M8
G0 X171.51 Y-34.833
Z52`

andrewmarles, 2022-01-31:
Additional info, this is using websockets and VFD control is enabled.

terjeio, 2022-01-31:
Looks lik this is due to a contoller regression, fix just commited.

andrewmarles, 2022-02-01:
Looks good, issue appears resolved with the latest commit.  Aggressive buffering working well.  Thanks so much for the quick response.

closed 2022-02-01 by andrewmarles