terjeio/ioSender #200: Positioning camera controll

calabr, 2022-02-12:
Trying to play with camera window.
Sender 2.0.37p1 
  I realized, the camera support is only few initial features implemented, but ants to report some issue for future.
1) selecting different camera didn't work - I connected USB endoscopic camera in addition to embedder notebook camera.
   Sender show both camera in the list, always selects  only first in the list. Trying to switch camera change camera name in the listbox, but don't switch image.  Zoom can use both cameras, so it's isn't the driver issue.
 Selected camera don't remembered between program launch.
 After camera window closed camera still enabled and selected by the app.
 after sender closed camera still enabled sometimes randomly. 
2) Camera "move offset" button executed second time, try to move again the same offset. Logically it should remember offset state and not allow to offset again, move back on second press.
  As I understand, it's no way currently to use camera for probing and correction with camera offset.

terjeio, 2022-02-12:
> selecting different camera didn't work...

Some bugs to sort out...

> Camera "move offset" button executed second time, try to move again the same offset. Logically it should remember offset state and not allow to offset again, move back on second press.

The camera window has to be closed and reopened to move the offset again? Or reenable the move offset button after jogging? Something else? IIRC the Mach3 camera plugin I use just moves the offset again.

> As I understand, it's no way currently to use camera for probing and correction with camera offset.

Why not? After moving the offset you can set the XY position to whatever you want, including zeroing. Or do you mean using the camera instead of a probe in the _Edge finder_ tabs? Like in the _Rotation_ tab?

calabr, 2022-02-13:
> The camera window has to be closed and reopened to move the offset again? Or reenable the move offset button after jogging? Something else? IIRC the Mach3 camera plugin I use just moves the offset again.

I think, offset should set the real offset, like G92. After this offset set first time, it should move position to look at spindle point, but work coordinates still the same. If I then move somewhere and set 0, then unset offset, working system zero become at the true 0 of workpiece.
  For re-enabling offset 
   1) after offset button pressed, it should change behavior to  "undo offset". If user close camera window with offset - rise the warning and ask user to undo offset, or leave it applied. Remember camera offset state in the global variable. Add button to reset offset state without any changes ( just in case user reset it manually before open the window)
 For safety rezone, add check-box "move to offset" - user can decide to just set/unset the coordinates, or perform any move according to it.
  

> Why not? After moving the offset you can set the XY position to whatever you want, including zeroing. Or do you mean using the camera instead of a probe in the Edge finder tabs? Like in the Rotation tab?

Currently it don't use offset for zero calculation.
Yes I want to use camera instead of probe.

It's also something wrong with move offset set to X-10 Y-30, but move is the next
```
<Idle|MPos:-95.373,-90.940,0.000|Bf:15,128|FS:0,0>
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
X10.000Y30.000
<Run|MPos:-95.320,-90.783,0.000|Bf:13,128|FS:3036,0>
<Hold:1|MPos:-93.548,-85.468,0.000|Bf:12,53|FS:780,0>
<Hold:1|MPos:-93.095,-84.108,0.000|Bf:12,0|FS:0,0>
ALARM:2 
```

terjeio, 2022-02-13:
I have implemented the same workflow as the Mach3 plugin uses. _Move offset_ moves the spindle to the position the camera looks at. To set the initial camera offset set XY to 0 and drill a small hole in something. Then locate this hole with the camera, open the _Settings: App tab_ and press _Get current position_ in the _Camera_ box and save the settings. I usually verify that the offset is correct by then clicking _Move offset_ in the camera window and check that the drill fits exactly into the hole.

I use the camera to locate the point on the workpiece where I want the tool to be, press _Move offset_ and set the current position by either entering values in the DRO or by zeroing X and Y.
Using the camera instead of a probe as a edge finder is thus easy, IMO there is no need to add a camera option to the _Edge finder_ tabs. I use the camera regularly to locate edges on my Mach3 equipped router and I am happy with the simple workflow.
The camera could be used instead of a probe for the _Center finder_ though. I'll look into that.

For those who want to add a camera: a tricky part of setting up a camera is getting the camera axis parallel to the Z axis...

calabr, 2022-02-14:
So, you use offset in the opposite direction I mansion.
Why not to set two buttons - move offset forward = set camera to the spindle position.
- move offset back = revert spindle back to camera position. 
It will allow locate reference point, then set spindle to this position, than quickly switch to camera again without manual search for position and only fine-tune it if needed.
  Also, if program aware of current positioning mode (camera or spindle) it can use offset to shift zero point or any coordinate set automatically. I think it will be easier to use. Especially because real camera offset will not be like 30.0 - it will be like 29.316 .
  currently, camera can be used for center-finder of round objects manually - just align of red ring with round edge, but it's hard to use for different shape of objects.
 Also it will be good to use it for rotation correction.
Allowing to use camera will utilize the same workflow, as for finding the edge with cutter touch of workpiece, so this method will become available automatically.

terjeio, 2022-02-19:
> Why not to set two buttons - move offset forward = set camera to the spindle position.

I have added a checkbox for reversed motion in the [latest edge version](https://github.com/terjeio/ioSender/discussions/176#discussioncomment-2211985).

> Also, if program aware of current positioning mode (camera or spindle) it can use offset to shift zero point or any coordinate set automatically. I think it will be easier to use.

I do not understand what you mean by this. Example please.

> Allowing to use camera will utilize the same workflow, as for finding the edge with cutter touch of workpiece, so this method will become available automatically.

I have added a button in the camera window for posting the current position as a probe event, and a button in the center finder  tab for using the positions provided by the camera. Experimental for now and not tested much. No input validation except for number of probe events.

calabr, 2022-02-21:
> I have added a checkbox for reversed motion in the https://github.com/terjeio/ioSender/discussions/176#discussioncomment-2211985.

Trying to play with 27p2. Without CNC - just notebook camera, locating my nose.
Checkbox meaning not very clear at first glance.
Offset button don't work in probe mode.
This checkbox seems not very practical for me - If I need to move back - I should click checkbox, than offset button. Revert camera back - again 2 clicks. It's easy to forget switch checkbox and offset in wrong direction.
  I propose to restrict offset move to only once - a) first push to the button move camera offset, second push move it back. Button caption should be changed to correspond with button action.
   b) use 2 buttons - "go camera" and "back to spindle" Only 1 button should be active at the time.
 If need to move offset again - close and open camera window.

> I do not understand what you mean by this. Example please.
   Workflow like this:     I want to zero at the center of the hole.
   1) position spindle near the hole
   2) open camera window, press offset. Now camera can see hole.
   3) use jogging to align camera with the hole.
   4a) press "Zero X" Zero Y" Sender know the offset of camera and know I'm already applied camera offset. It add this offset to the G10L2 command, so spindle position will become X0 Y0.
    4b) revert offset, than press "Zero X" Zero Y" Sender know the offset not applied and set current position as X0 Y0.
 The same behavior can be used for probe operation with camera.

> I have added a button in the camera window for posting the current position as a probe event, and a button in the center finder tab for using the positions provided by the camera.

 It's good idea to record camera positions list and than use it for probe!
 Some suggestions to make it more useful - recorded positions should be seen on the probe page. At the begging - empty table with string numbers, corresponding to test point number on the picture. It will be grate to highlight next probing point on the picture. 
  If all strings filled - first string become active.
  When user select any string in the table, it's become next active point and next capture of camera position will fill this string. So, user can redo measure in case of mistake.
  "Capture position" button should be located on Probe window. The same button will allow fully manual position set - by touching the edge with cutter.
   Filling this table during normal probe procedure will better visualize probe process.
   Also you can add checkbox to capture-only - so probe procedure will fill-in the table, than user can decide to apply it or not.

   Opening camera window starts windows camera service, but closing it didn't stop. As result, camera still working until application will be closed.
  Windows camera frame server still running and eat 8% of CPU on not so slow I7 processor. I don't check it on old notebook for my CNC yet, but seems it can seriously slow-down it.
[image]

terjeio, 2022-02-21:
>  Only 1 button should be active at the time.

Which one to set active on open? Add a checkbox for that or ask the user via a message box? Or alternate between openings?
For my main workflow I want to "move spindle to camera position" to be the default.

I'll have ty try your 1-4 step procedure above so no comment on that yet.

>  recorded positions should be seen on the probe page.

I'll add them to the preview list, hiding the preview clears the captured positions. Capturing a position (re)opens it. 

> When user select any string in the table, it's become next active point and next capture of camera position will fill this string. So, user can redo measure in case of mistake.

I am lazy to code this, if a mistake has been made clear the captured positions and start over.
If you want this badly you can give it a try? By overlaying a new control over the preview?

> "Capture position" button should be located on Probe window. The same button will allow fully manual position set - by touching the edge with cutter.

I'll consider this later - the camera handling is in a separate project and I do not want to drag offset handling into the probing code. BTW it works for the _Rotation_ tab as that uses the relative distance between two positions to determine the angle.

> Also you can add checkbox to capture-only - so probe procedure will fill-in the table, than user can decide to apply it or not.

I still need the "Use camera positions" button to apply the captured positions so IMO a checkbox is not needed. Captured positions are cleared on a probing tab change or by hiding the preview.

terjeio, 2022-02-21:
[New edge version](https://github.com/terjeio/ioSender/discussions/176#discussioncomment-2220757) is now up.

Ref. 4a and 4b above, why do you want to set the offset two times by zeroing X and Y?

calabr, 2022-02-28:
Sorry for late reply, but the war made some correction to my plans.
> Which one to set active on open? Add a checkbox for that or ask the user via a message box? Or alternate between openings?
 For my workflow - "move camera to spindle position" is more suitable.
 Checkbox is more appropriate.
 
> I am lazy to code this, if a mistake has been made clear the captured positions and start over.

what is the behavior if user captured more points, than needed for current probe type? use first set, last one?

> By overlaying a new control over the preview?
  I think, it should be separated list - you can also fill it with coordinates during normal probe operation - without camera.
 If it's not enough space for 2 lists - preview can be opened as separate window. This information is only needed during probe setup and troubleshooting.
 
> I still need the "Use camera positions" button to apply the captured positions so IMO a checkbox is not needed. 
I mention not the camera bode, but probing with hardware probe.

> Ref. 4a and 4b above, why do you want to set the offset two times by zeroing X and Y?
  If it's allowed to close window DON'T reposition back to shpindle instead of camera -
  ___user can move it back manually and need to apply offset again

calabr, 2022-02-28:
Sorry for late reply, but the war made some correction to my plans.

> Which one to set active on open? Add a checkbox for that or ask the user via a message box? Or alternate between openings?

For my workflow - "move camera to spindle position" is more suitable.
Checkbox is more appropriate.

> I am lazy to code this, if a mistake has been made clear the captured positions and start over.

what is the behavior if user captured more points, than needed for current probe type? use first set, last one?

> By overlaying a new control over the preview?

I think, it should be separated list - you can also fill it with coordinates during normal probe operation - without camera.
If it's not enough space for 2 lists - preview can be opened as separate window. This information is only needed during probe setup and troubleshooting.

> I still need the "Use camera positions" button to apply the captured positions so IMO a checkbox is not needed.

I mention not the camera mode, but probing with hardware probe.

> Ref. 4a and 4b above, why do you want to set the offset two times by zeroing X and Y?

If it's allowed to close window without repositioning back to spindle instead of camera - user can move it back manually and need to apply offset again

terjeio, 2022-03-02:
> For my workflow - "move camera to spindle position" is more suitable.

I have added a checkbox to camera options in _Settings: App_ for default action when camera window is opened. The latest edge versions with this has just been uploaded.

>  This information is only needed during probe setup and troubleshooting.

Opening the console window should have the needed info, _File > Open Console_. If not I can add the missing bits there.

> If it's allowed to close window without repositioning back to spindle instead of camera - user can move it back manually and need to apply offset again

It is allowed as the camera can be used for measurements as well, I do that for precisely locating holes/features that are otherwise hard to determine (e.g. with a caliper).

calabr, 2022-04-15:
2.0.38p1
> I have added a checkbox to camera options in _Settings: App_ for default action when camera window is opened. The latest edge versions with this has just been uploaded.

Checkbox in settings works not as described - it's disable/enable "Move offset" button
When enabled, Checkbox " camera to spindle" set on window open, cleared on Move offset button press, but if you press this button again, it's not set " camera to spindle" checkbox and move the same direction again.
  Intuitively, it should invert checkbox state at every button press.

As I remember, you made some changes to camera initialize/close processing, but in this version camera still works in background when window closed until application close.

terjeio, 2022-04-20:
Fixed in [latest edge version](http://www.io-engineering.com/downloads)?

Note that _Move offset_ button is only enabled when the `Grbl` tab is active.

calabr, 2022-04-21:
> Fixed in [latest edge version](http://www.io-engineering.com/downloads)?
Note that Move offset button is only enabled when the Grbl tab is active.

Looks like now it's Ok
Probable it's no need to block offsetting in probe mode.

closed 2022-04-27 by calabr