terjeio/ioSender #204: ioSender does not handle some Grbl crashes cleanly

phil-barrett, 2022-02-16:
We are using ioSender with a Grbl "classic" (1.1h, Arduino Mega) and get a series of "Unhandled exception" error messages when the Arduino Mega crashes.  We have to then go into task manager and kill off ioSender.  Windows 10, not sure what update level.  Here is an example.
[image]

The Mega is crashing due to EMI (what fun!) which I am fixing - part of which is tossing the Mega and replacing it with a T41U5XBB.  But, at the least, we should be able to terminate ioSender without the task manager. Our users are not very sophisticated and blame ioSender rather than the crappy MillRight electronics (no filtering/isolation on any of the inputs, no shielding on wires, acrylic electronics case cover).

terjeio, 2022-02-21:
[New edge version](https://github.com/terjeio/ioSender/discussions/176#discussioncomment-2220757) is now up with some changes that hopefully helps.

>  I find it sad that companies ship products that completely ignore the problems with EMI.

I believe this is common for hardware designed for legacy Grbl, make it as cheap as possible and leave issues for the buyer to sort out - or blame it on bad code?

phil-barrett, 2022-02-21:
Thanks.  I will try to test it out with the grbl board though that environment is no longer.  Upgraded the MillRight Mega V to grblHAL on the T41U5XBB.  No more crashes!  And I've got a bunch of believers at the makerspace!
[Upgrade is described here.](https://www.grbl.org/single-post/millright-mega-v-brain-transplant)

closed 2023-04-29 by terjeio