terjeio/ioSender #216: Connection to PI Pico / RP2040

oke24, 2022-04-02:
I try to connect ioSender on a Windows 10 PC to a PI Pico with grblhal with RP2040 driver.
The setup as shown with CNCjs on a PI Zero 2 W and PIPico on a PicoBOB works.

Now I try to use ioSender on a Windows PC because of the extented capabilities compared to CNCjs.  
But I'm not able to connect ioSender to the PiPico board. When connecting PiPico with a USB cable to the PC there is no new USB device  resp. COM-Port available to connect to (only in bootsel mode I get an USB drive device).

Alternatively I have the option to connect ioSender on Windows PC via IP / WLAN to the Pi Zero Board and connect from there to the Pi Pico.  

I would be glad to get some hints how to connect ioSender, or if this is not possible.

[image]

andrewmarles, 2022-04-02:
You can connect the Windows 10 PC directly to the Picobob with USB.  Or else you can use SER2NET in linux to connect the USB serial port on the Pi Zero to the wifi and then point IOSender at that address and port.  One of the members on the PrintNC discord has done this and it worked well.

oke24, 2022-04-02:
@andrewmarles , thank you very much for the hints. I have a look at SER2NET.

In the meantime I tried a websocket-serial bridge, found on Github under https://github.com/platisd/SocketSerialBridge .
This will be a very lightweight solution for my problem, but it doesn't run.

The bridge is running on the PI Zero but ioSender can't connect via Websocket on Port 8088 with a timeout after some time.
After initial communication with the bridge (some headers) ioSender reports "no websocket connection on ws://192.168.1.247:8088".

[image]

phil-barrett, 2022-04-02:
Web socket port default is 80.

oke24, 2022-04-02:
yes, but as I understood in ioSender I can specify a different WS Port for communication. In my example above I do this on the Windows command line to port 8088 with the "-port" option, where the Serial Bridge application on the PI zero waits for connection.

terjeio, 2022-04-02:
> But I'm not able to connect ioSender to the PiPico board. When connecting PiPico with a USB cable to the PC there is no new USB device resp. COM-Port available to connect to (only in bootsel mode I get an USB drive device).

Win10 driver should install automatically (it did so for me on a brand new machine) so a bit strange if you built with USB_SERIAL_CDC enabled.

The initial websocket handshake seems to work, but then it fails? I am using a 3rd party WebSocket library for ioSender and I do not know if there is any issues with that. To be investigated?

phil-barrett, 2022-04-02:
But doesn't grblHAL need to know what socket to connect to?  The default in grblHAL is 80. Have you changed it in Grbl Settings?
[image]

oke24, 2022-04-03:
> 
> Win10 driver should install automatically (it did so for me on a brand new machine) so a bit strange if you built with USB_SERIAL_CDC enabled.

Yes, USB_SERIAL_CDC is enabled in my_machine.h:
```
#ifndef USB_SERIAL_CDC
#define USB_SERIAL_CDC       1 // Serial communication via native USB.
#endif
```
But there is unfortunately no new COM-Port when connecting to Windows 10 PC. 
> 
> The initial websocket handshake seems to work, but then it fails? I am using a 3rd party WebSocket library for ioSender and I do not know if there is any issues with that. To be investigated?

Yes, the initial handshake is always there when starting ioSender via cmd-line on port 8088. But then communication stops immediately and after 80 sec estimated the connection was closed by ioSender. As shown in the screenshot above on Windows PC I got the error message and the bridge on the Pi released the connection and opend the port again for next connection.

oke24, 2022-04-03:
> But doesn't grblHAL need to know what socket to connect to? The default in grblHAL is 80. Have you changed it in Grbl Settings?

In ioSender the initial connection is asked when starting the app and no App.config file is available:

[image]

Result is the same as when starting ioSender on the command line 
`iosender -port ws://192.168.1.247:8088`

Websocket communication is between PC and Pi Zero, where the websocket-serial bridge is waiting for connection. The bridge connects on the other side to the serial port of the Pi Pico.

phil-barrett, 2022-04-03:
This is what I get. Have you actually tried port 80?
[image]
It boots and I get this.
[image]

oke24, 2022-04-03:
@phil-barrett ; @terjeio 
Hi, thank you very much for further investigating …

Port 80 isn’t actually available on my Pi Zero because I have a listener on this port. 

High ports are OK, so I use 8088 for the Websocket test, e.g. CNCjs runs on port 8000 as default without a problem. 
I think the problem should‘nt be dependent on the port chosen, or maybe there is a hint for this ?

Best regards

terjeio, 2022-04-03:
@oke24 I'll look into this a bit later as I am busy with a project in my garden that I want to complete before easter.

oke24, 2022-04-04:
@terjeio, fine - thank you very much for your efforts and good luck !

oke24, 2022-04-04:
@andrewmarles , terjeio will take a look at the websocket communication problem. 
In the meantime as you mentioned I installed SER2NET on the PI Zero. Very straightforward. I'm able to connect to the new established telnet port and communicate with the PI Pico bridged by PI Zero. So I have a solution connecting ioSender with PicoBOB via telnet.

terjeio, 2022-04-21:
> In the meantime I tried a websocket-serial bridge

I have had a look at the code and I seems to be a plain socket server and thus _could_ be usable for a telnet connection. The welcome message should be edited out, and the code has to be modified for byte transfers as it supports strings only?
From the readme:
>  It relays messages as strings from the websocket connection to the serial and vice versa. You can easily modify this to handle bytes.

terjeio, 2022-09-24:
FYI [Pi Pico W](https://github.com/grblHAL/RP2040) can now be used as a direct replacement for Pi Pico and has native support for all grblHAL networking protocols, and WebUI v3, over WiFi.

closed 2022-09-24 by terjeio