terjeio/ioSender #217: When building a surface map, the controller hangs accidentally

Maksko, 2022-04-05:
When building a surface map, the controller hangs accidentally. The CurrentDomainException window appears: "Unhandled exception occured". Program version Release 
ioSender 2.0.37. Makerbase MKS DLC32 controller, FluidNC firmware v3.4.2. Wifi connection.
Where do I look for error?

terjeio, 2022-04-20:
I have run a few thousand probing actions over USB with no issues. Can you try with the [latest edge version](http://www.io-engineering.com/downloads).

terjeio, 2022-05-09:
The latest [edge versions](http://www.io-engineering.com/downloads) has been improved to tolerate more badly formed/corrupted reports from the controller. They will silently be ignored.

closed 2023-04-29 by terjeio