terjeio/ioSender #218: ioSender stays greyed out after MPG:1->MPG:0 switch

stevempotter, 2022-04-08:
My wireless pendant programming is progressing well. I am using hardware switching (which I set to Teensy pin 41) to allow my pendant to take control of the CNC. Generally, ioSender is well behaved, going grey and being unresponsive while grbl is in Mode:1 and the pendant is sending jog commands. But ioSender frequently gets confused and stays greyed-out after the mode is switched back to MPG:0. I attach a screenshot showing this confused and unresponsive state. I am using V. 2.0.36p14 of ioSender. Let me know if you need additional info to help debug this, for more robust usage of MPGs and pendants in general.

terjeio, 2022-06-25:
I believe I've finally managed to sort this out. Can you try with the [latest edge version](http://www.io-engineering.com/downloads) (2.0.39p1).

stevempotter, 2022-08-02:
I have been busy doing other things since you released 2.0.39p1... Now I am getting back to my pendant project. I have just downloaded and installed 2.0.40, and copied in my settings.txt and App.config files into its folder. When I try to run it, a window pops up saying "MPG mode is active, waiting for it to be disabled." but my pendant is off and the NOT_MPG mode line going into the Teensy pin 41 is high, as it should be when I am not using MPG control. In my_machine.h, I have set MPG_ENABLE 2 and KEYPAD_ENABLE 2.

Please let me know if I need to choose some other settings in my grblHAL firmware or wire things up differently to get back to using ioSender.

Thanks!

stevempotter, 2022-08-02:
BTW, I have 
#define MPG_MODE_PIN        (41u)
in TT41U5XBB_map.h line 68
but I don't have I2C_STROBE_ENABLE set since I am using UART control, not I2C.
Is that right?

stevempotter, 2022-08-02:
I tried installing 2.0.39 and it did start up OK with the pendant off. I turned on the pendant and ioSender goes grey as expected, and I have MPG control. But when I turn off the pendant, ioSender stays greyed out and I can't do anything with it but End Task in Task Manager. Please let me know what other information I can provide to help debug this.

stevempotter, 2022-08-03:
I think ioSender may be getting overwhelmed listening in on the status reports that my pendant is requesting for every jog move. It has no problem taking back control when I turn off the pendant as long as I don't do any jogging. But if I do some jogging, with a few hundred joglets, when I turn off the pendant, ioSender never stops being greyed out. It is completely stuck, unable to click on anything in ioSender or even close the window. I can only halt ioSender using Task Manager.

grblHAL seems to be working fine. I can turn the pendant on and off and jog with it when ioSender is locked up in a greyed-out state.

stevempotter, 2022-08-05:
OK, I verified that the problem is ioSender's Console window getting overloaded with status reports during pendant jogging. If I never use the console window, or if I use it but don't tick the Verbose box, then the MPG mode switching works fine. Only when I tick Verbose and do some jogging does ioSender get locked up (stays greyed out) and unable to recover without hitting End Task in Task Manager.
It is nice to be able to see status reports in the Console when jogging but not absolutely needed. 
I use those to debug and optimize jogging algorithms.
But it would be good to have a better way to get out of a locked up ioSender than End Task.