terjeio/ioSender #219: Problems with offline use of software

lianzaozi, 2022-04-10:
Open the software without connecting the hardware, the following window will pop up, click the OK button, and the software will exit. This will cause the software to be unavailable when offline.

[image]

phil-barrett, 2022-04-10:
This is a known issue - ioSender needs to be connected to an actual grblHAL device.  

To be honest, I do not think this is a terrible "feature" as I have seen so many problem posts about other GCode Senders that came from not being connected. 

There is a minor bug in the wording of that dialog, though.  Response is misspelled.

lianzaozi, 2022-04-10:
Will the offline feature be added in the next version?

terjeio, 2022-04-10:
> Will the offline feature be added in the next version?

Not likely, as the sender configures itself at startup from data read from the controller. Changing this will require a fair bit of work.

What would the use cases for running it offline be? Viewing gcode in the 3D viewer?

lianzaozi, 2022-04-13:
The offline function is usually used to preview the gcode, simulate the gcode, and check whether the movement of the gcode trajectory is correct. This way you can avoid some mistakes before real machining.
Offline functions such as laser grbl and grbl plotter software.

terjeio, 2022-04-20:
> There is a minor bug in the wording of that dialog, though. Response is misspelled.

Fixed in [latest edge version](http://www.io-engineering.com/downloads).

kimstik, 2022-07-28:
IMHO offline mode is essential.