terjeio/ioSender #224: Documentation: Probing

oke24, 2022-05-17:
I'm looking for a good in depth description of the many options for probing but can't find a complete documentation but only special aspects in probing. Currently this is a trial and error process for me.

[image]

So can anyone provide me with a good link to a probing documentation with iosender and also how probing processes should be done in the tab "Tool lenght offset", e.g.  when the starting point of the spindle is not the home position but has an xy-offset. Thank you very much! I'm also very interested in the description of the options on the other tabs.

S2000Stefan, 2022-05-18:
Hello oke24,
I have started to make some tutorials here in order to show the full range of functions of the ingenious programme ioSender in connection with grblHAL. 
Unfortunately, I am very busy at the moment and can only make a few or no videos on this topic. I hope that will change soon. 
Maybe my videos will help you a bit.
Here is the link:
https://www.youtube.com/playlist?list=PLnSV6o2cRxM5mQQe4ec5cS2J8jBsEciY3

oke24, 2022-05-18:
Hi S2000Stefan, thank you for reply. I found your very useful videos on YT. And I would be glad if you find some time to complete the videos. Specially the first tab "tool length offset" will be of interest because I'm not really sure what each option does.