terjeio/ioSender #226: missing signals LED & 3D View offset handling

oke24, 2022-05-20:
Hi, when working with iosender I have 2 more questions regarding the GUI. 

First I don't see any signal led's even if routing is started. My understanding is that I should see minimum a red S-led signal. 

Second problem is that I'm not able to position the red tool animation dot in 3D view on the right place. I set an X-offset as shown in the picture to place the tool above the workpiece, do a G54 offset position setting, but the dot is always the amout of x-offset on the right side. I also recognize that editing the DRO to "zero" will not work. The values in x,y,z reamain at the positions shown in the picture. How can I place the dot to the right position ? Thank you very much !

[image]

terjeio, 2022-05-20:
> First I don't see any signal led's even if routing is started. My understanding is that I should see minimum a red S-led signal.

Normally none of the signal LEDs should be lit. They reflect the status of the input pins and the S is the cycle start input (hover the mouse over the LED to see the full description). Cycle start and feed hold inputs are typically connected to momentary switches, if at all. Do you have them connected?

>  I set an X-offset as shown in the picture to place the tool above the workpiece, do a G54 offset position setting, but the dot is always the amout of x-offset on the right side.

Do you have a grblHAL controller? If so is _Work coordinate offset_ checked for the $10 setting?

[image]

> I also recognize that editing the DRO to "zero" will not work.

You have to press the [enter key](https://github.com/terjeio/ioSender/wiki/Usage-tips#setting-current-position) to set the value.

oke24, 2022-05-21:
> Normally none of the signal LEDs should be lit. They reflect the status of the input pins and the S is the cycle start input (hover the mouse over the LED to see the full description). Cycle start [...]
Thank you very much for explanation, the meaning has now become clear to me. I see LEDs red, e.g. at the endstops. 
 
> Do you have a grblHAL controller? If so is _Work coordinate offset_ checked for the $10 setting?
>
> [image]
>
>
> You have to press the [enter [...]

The settings above solved the problem. Thank your very much for this hint. 

Lastly, I would be very grateful if a documentation about the many options of probing would be available, especially about the tab "tool length offset". Currently the usage is more random for me and I'm not sure how to safely apply the options when probing with offset values.  #224