terjeio/ioSender #228: IOSender 2.0.37 Tooloffset, 2 bugs?

Team-Deepflights, 2022-05-29:
Btw, it found a few "bugs" in IOSender 2.0.37:
I tried C&P the X Z values into the tooloffset, the values are there but if I click the button to save them, nothing happens. I had to retype them manually.
If a tool with offsets is loaded, the value is gone in the offset page if you click on "Reset", it´s lost at all. also rebootig keeps the affected toolfield to 0,00 0,00. ( I was lucky, I copied the values into a TXT) :-)

terjeio, 2022-05-31:
> I tried C&P the X Z values into the tooloffset, the values are there but if I click the button to save them, nothing happens. I had to retype them manually.

Fixed in the latest [edge version ](http://www.io-engineering.com/downloads).

> If a tool with offsets is loaded, the value is gone in the offset page if you click on "Reset", it´s lost at all.

I am a bit usure about what you mean by this... Do you have a controller with a tooltable compiled in or are you using the generic offsets (G5x)? 

>  also rebootig keeps the affected toolfield to 0,00 0,00.

A reboot clears current tool and active tool offset. M61 can be used to reestablish the current tool. The offset has to be reestablished as well either from the tooltable or explicitly.

Team-Deepflights, 2022-05-31:
The tooltable is compiled because I have more different tools (18) as the usual G5x offsets.
I use M61 in a homingmacro to set tool 0 for reference. 
Call M61 is worthless if the tooltable have no offsets set. :-)

I tell you that happened, please remember we are talking about a lathe, the gcode load tool 16, but while going G0 to the starting point, I realize that the material to be processed is not at ZO so I hit reset the lathe stops as it should.
I know, cycle stop would have been better, but G0 in my lathe is really fast, there was no time for a decision :-)
So I home again, set the material to Z0 and start the cycle again, the lathe go to do the toolchange and wait (manual toolchange). There I saw that the coordinates did not change and so I took a look at the tooltable, tool 16 was 0,000 in both axis.

I tried this 2 times with different toolstation numbers to replicate, could have also been my mistake, on both the called tool was erased after stopping with emergency stop.

terjeio, 2022-06-01:
There are bugs related to this, change [this code](https://github.com/grblHAL/core/blob/6d00f743f09db2a07adadf4dc996e3d55de37cdb/gcode.c#L213-L230) to

```
        case ToolLengthOffset_Cancel:
            idx = N_AXIS;
            do {
                idx--;
                tlo_changed |= gc_state.tool_length_offset[idx] != 0.0f;
                gc_state.tool_length_offset[idx] = 0.0f;
#ifndef N_TOOLS
                gc_state.tool->offset[idx] = 0.0f;
#endif
            } while(idx);
            break;

        case ToolLengthOffset_EnableDynamic:
            {
                float new_offset = offset / settings.axis[idx].steps_per_mm;
                tlo_changed |= gc_state.tool_length_offset[idx] != new_offset;
                gc_state.tool_length_offset[idx] = new_offset;
#ifndef N_TOOLS
                gc_state.tool->offset[idx] = offset;
#endif
            }
            break;
```

and try again. I will look into this in more detail a bit later since there is a todo comment in the code about whether the offset should be kept over a soft reset/e-stop when a tool table is compiled in. Do you know how other controllers handle this?

closed 2022-06-13 by terjeio