terjeio/ioSender #230: COM port error

tetref, 2022-06-16:
Probably since version 2.0.37, ioSender has a special error when creating a default configuration. After editing App.config, everything works. This may apply to specific Windows language versions ..
[image]
[image]

terjeio, 2022-08-16:
Can you try the latest [edge version](http://www.io-engineering.com/downloads)? It looks like there is a superfluous space after the reported com port name messing up.

tetref, 2022-08-23:
2.0.41p2 Edge - left picture NOT WORKS
2.0.41p2 XL Edge - right picture - WORKS

[image]

terjeio, 2022-09-15:
Looks like the update somehow went wrong, I have now uploaded a new edge version that should be ok.

tetref, 2022-09-16:
You can close this thread, both 2.0.41p3 versions are working now.

closed 2022-09-16 by terjeio