terjeio/ioSender #241: E-Stop Issue

silver2row, 2022-08-22:
Hello,

I am receiving an E-Stop issue. I am a novice w/ this software so far and I just wanted to test it for now.

Anyway, here is the photo of my current issue. It keeps popping up.

[image]

andrewmarles, 2022-08-22:
You need to either connect a switch and close the e-stop, or else you can use a terminal program to access the controller and invert the control signal such as described [here](https://www.grbl.org/single-post/naked-teensy-party).

phil-barrett, 2022-08-22:
Yes, ioSender won't start up with the E-Stop active. 

Another place to read up on this is in the [First Run wiki ](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings) section.

silver2row, 2022-08-23:
Hello @andrewmarles and @phil-barrett ,

Thank you guys. I will read more about the source soon. 

Seth

closed 2022-08-23 by silver2row