terjeio/ioSender #242: inner center finder multi-path error

calabr, 2022-08-24:
sender 2.0.41p2
 probbing -> center finder->  inner center finder
 using more then 1 pass  cause probbing error or trow exception and close the program.
 

```
**DispatcherException

Unhandled exception occured: Exception has been thrown by the target of an invocation.**
```

Here is result of probbing with 1 pass - successfully several times in the row
then I changed to 2 passes without changing any setup. First pass finished successfully, second one trow exception in the begginig.
Sometimes it's run 2 passes successfully, than error.

Outer center probe didn't show such error
```
G53G1F2000X-260.815
PM:ok
G38.2F300Y-30
<Run|MPos:-255.043,-160.540,-80.158|Bf:14,128|FS:1680,0>
<Run|MPos:-260.260,-160.540,-80.158|Bf:14,128|FS:219,0>
<Run|MPos:-260.815,-160.918,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.815,-161.543,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.815,-162.165,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.815,-162.865,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.815,-163.490,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.815,-164.115,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.815,-164.813,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.815,-165.438,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.815,-166.063,-80.158|Bf:14,128|FS:300,0>
[PRB:-260.815,-166.358,-80.158:1]
<Idle|MPos:-260.815,-166.638,-80.158|Bf:15,128|FS:0,0|Pn:P|Ov:100,100,100>
PM:ok
G1F2000Y0.5
PM:ok
G38.2F25Y-2
<Run|MPos:-260.815,-166.145,-80.158|Bf:15,128|FS:0,0>
<Run|MPos:-260.815,-166.190,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-260.815,-166.245,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-260.815,-166.298,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-260.815,-166.348,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-260.815,-166.408,-80.158|Bf:14,128|FS:25,0>
[PRB:-260.815,-166.415,-80.158:1]
<Idle|MPos:-260.815,-166.438,-80.158|Bf:15,128|FS:0,0|Pn:P>
PM:ok
G53G1F2000Y-160.54
PM:ok
G38.2F300Y30
<Run|MPos:-260.815,-165.323,-80.158|Bf:14,128|FS:2000,0>
<Run|MPos:-260.815,-160.968,-80.158|Bf:14,128|FS:6,0>
<Run|MPos:-260.815,-160.185,-80.158|Bf:14,128|FS:300,0|Ov:100,100,100>
<Run|MPos:-260.815,-159.560,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.815,-158.863,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.815,-158.238,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.815,-157.613,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.815,-156.913,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.815,-156.288,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.815,-155.665,-80.158|Bf:14,128|FS:300,0|WCO:-260.815,-160.539,-80.360>
<Run|MPos:-260.815,-154.965,-80.158|Bf:14,128|FS:300,0>
[PRB:-260.815,-154.740,-80.158:1]
<Idle|MPos:-260.815,-154.492,-80.158|Bf:15,128|FS:0,0|Pn:P>
PM:ok
G1F2000Y-0.5
PM:ok
G38.2F25Y2
<Run|MPos:-260.815,-154.965,-80.158|Bf:15,128|FS:0,0>
<Run|MPos:-260.815,-154.940,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-260.815,-154.888,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-260.815,-154.837,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-260.815,-154.778,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-260.815,-154.725,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-260.815,-154.675,-80.158|Bf:14,128|FS:25,0|Pn:P>
[PRB:-260.815,-154.675,-80.158:1]
PM:ok
G53G1F2000Y-160.54
PM:ok
G53G1F2000Z-75.158
<Run|MPos:-260.815,-154.815,-80.158|Bf:14,128|FS:1680,0>
PM:ok
G53G1F2000X-260.812Y-160.545
<Run|MPos:-260.815,-156.773,-80.158|Bf:12,128|FS:2000,0>
<Run|MPos:-260.815,-160.540,-80.145|Bf:13,128|FS:1422,0>
<Run|MPos:-260.815,-160.540,-76.545|Bf:13,128|FS:835,0|Ov:100,100,100>
<Idle|MPos:-260.813,-160.545,-75.158|Bf:15,128|FS:0,0>
G92X0Y0
G92X0Y0
$G
G91
Probing completed: X distance 13.875, Y distance 13.94.
[GC:G1 G54 G17 G21 G91 G94 M5 M9 M56 T0 F2000 S0]
<Idle|MPos:-260.813,-160.545,-75.158|Bf:15,128|FS:0,0|WCO:-260.812,-160.545,-80.360>
<Idle|MPos:-260.813,-160.545,-75.158|Bf:15,128|FS:0,0|Ov:100,100,100>

```

```
Probing, pass 1 of 2...
G91F300
PM:ok
G53G1F2000Z-80.158
PM:ok
G38.2F300X-30
<Run|MPos:-260.813,-160.545,-79.243|Bf:14,128|FS:560,0>
<Run|MPos:-261.040,-160.545,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-261.725,-160.545,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-262.348,-160.545,-80.158|Bf:14,128|FS:300,0|Ov:100,100,100>
<Run|MPos:-262.973,-160.545,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-263.672,-160.545,-80.158|Bf:14,128|FS:300,0|WCO:-260.812,-160.545,-80.360>
<Run|MPos:-264.297,-160.545,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-264.922,-160.545,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-265.620,-160.545,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-266.245,-160.545,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-266.855,-160.545,-80.158|Bf:14,128|FS:0,0|Pn:P>
[PRB:-266.625,-160.545,-80.158:1]
PM:ok
G1F2000X0.5
PM:ok
G38.2F25X-2
<Run|MPos:-266.380,-160.545,-80.158|Bf:15,128|FS:0,0>
<Run|MPos:-266.408,-160.545,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-266.460,-160.545,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-266.513,-160.545,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-266.565,-160.545,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-266.618,-160.545,-80.158|Bf:14,128|FS:25,0>
[PRB:-266.650,-160.545,-80.158:1]
<Idle|MPos:-266.672,-160.545,-80.158|Bf:15,128|FS:0,0|Pn:P>
PM:ok
G53G1F2000X-260.813
PM:ok
G38.2F300X30
<Run|MPos:-266.573,-160.545,-80.158|Bf:14,128|FS:1680,0>
<Run|MPos:-263.163,-160.545,-80.158|Bf:14,128|FS:1659,0>
<Run|MPos:-260.715,-160.545,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.090,-160.545,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-259.465,-160.545,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-258.765,-160.545,-80.158|Bf:14,128|FS:300,0|Ov:100,100,100>
<Run|MPos:-258.140,-160.545,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-257.517,-160.545,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-256.818,-160.545,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-256.193,-160.545,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-255.568,-160.545,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-254.868,-160.545,-80.158|Bf:14,128|FS:0,0|Pn:P>
[PRB:-254.985,-160.545,-80.158:1]
PM:ok
G1F2000X-0.5
PM:ok
G38.2F25X2
<Run|MPos:-254.953,-160.545,-80.158|Bf:15,128|FS:0,0|Pn:P>
<Run|MPos:-255.180,-160.545,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-255.120,-160.545,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-255.068,-160.545,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-255.018,-160.545,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-254.958,-160.545,-80.158|Bf:14,128|FS:0,0|Pn:P|WCO:-260.812,-160.545,-80.360>
[PRB:-254.965,-160.545,-80.158:1]
PM:ok
G53G1F2000X-260.813
PM:ok
G38.2F300Y-30
<Run|MPos:-255.103,-160.545,-80.158|Bf:14,128|FS:1680,0>
<Run|MPos:-258.648,-160.545,-80.158|Bf:14,128|FS:1425,0>
<Run|MPos:-260.813,-160.670,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.813,-161.295,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.813,-161.920,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.813,-162.620,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.813,-163.245,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.813,-163.870,-80.158|Bf:14,128|FS:300,0|Ov:100,100,100>
<Run|MPos:-260.813,-164.568,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.813,-165.193,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.813,-165.818,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.813,-166.518,-80.158|Bf:14,128|FS:0,0|Pn:P>
[PRB:-260.813,-166.358,-80.158:1]
PM:ok
G1F2000Y0.5
PM:ok
G38.2F25Y-2
<Run|MPos:-260.813,-166.493,-80.158|Bf:14,128|FS:17,0|Pn:P>
<Run|MPos:-260.813,-166.123,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-260.813,-166.183,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-260.813,-166.233,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-260.813,-166.285,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-260.813,-166.345,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-260.813,-166.398,-80.158|Bf:14,128|FS:25,0>
[PRB:-260.813,-166.408,-80.158:1]
PM:ok
G53G1F2000Y-160.545
<Idle|MPos:-260.813,-166.430,-80.158|Bf:15,128|FS:0,0|Pn:P>
PM:ok
G38.2F300Y30
<Run|MPos:-260.813,-163.680,-80.158|Bf:14,128|FS:2000,0>
<Run|MPos:-260.813,-160.575,-80.158|Bf:15,128|FS:0,0>
<Run|MPos:-260.813,-160.020,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.813,-159.370,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.813,-158.745,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.813,-158.120,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.813,-157.420,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.813,-156.798,-80.158|Bf:14,128|FS:300,0|Ov:100,100,100>
<Run|MPos:-260.813,-156.173,-80.158|Bf:14,128|FS:300,0>
<Run|MPos:-260.813,-155.473,-80.158|Bf:14,128|FS:300,0|WCO:-260.812,-160.545,-80.360>
<Run|MPos:-260.813,-154.848,-80.158|Bf:14,128|FS:300,0>
[PRB:-260.813,-154.735,-80.158:1]
<Idle|MPos:-260.813,-154.498,-80.158|Bf:15,128|FS:0,0|Pn:P>
PM:ok
G1F2000Y-0.5
PM:ok
G38.2F25Y2
<Run|MPos:-260.813,-154.990,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-260.813,-154.938,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-260.813,-154.885,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-260.813,-154.828,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-260.813,-154.775,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-260.813,-154.723,-80.158|Bf:14,128|FS:25,0>
<Run|MPos:-260.813,-154.665,-80.158|Bf:14,128|FS:0,0|Pn:P>
[PRB:-260.813,-154.675,-80.158:1]
PM:ok
G53G1F2000Y-160.545
PM:ok
G53G1F2000Z-75.158
<Run|MPos:-260.813,-154.835,-80.158|Bf:14,128|FS:1680,0>
PM:ok
G53G1F2000X-260.808Y-160.541
```

terjeio, 2022-09-15:
I have tested with up to 9 passes (the max) without getting any exceptions, however got some random _Init failed_ messages causing the UI to hang up - perhaps it is related. I have fixed the _Init failed_ "bug" in the [latest edge version](http://www.io-engineering.com/downloads), so with luck the exception may go away.

calabr, 2022-09-24:
Checked with 2.0.41p3
  issue solved.
found new behaviour - if Preview checked, start button generate command list, but pressing ALT+R starts real probbing.

terjeio, 2022-09-25:
>  pressing ALT+R starts real probbing.

Fixed in the [latest edge version](http://www.io-engineering.com/downloads).